openthread/openthread #1158: NextHop is removed when the link is lost.

hubertmis, 2017-01-16:
Let's say there are 3 routers in the mesh: A, B and C. Every router has established link with each other. When router A loses link to the router C it clears C's NextHop field. While it should set C's NextHop to router B (to forward frames to C via B).

jwhui, 2017-01-17:
I agree that it would be ideal for A to set C's NextHop field to B.  However, routers currently do not maintain the Route TLV received from all neighboring routers.  As a result, A will need to wait until the next MLE Advertisement from B before updating the route to C.

How were you thinking that A could update its route to C?

LuDuda, 2017-01-17:
I think it could be easily done if we follow the Thread Specification and have sort of "link set" and "route set" without maintaining Route TLV from all routers. What i mean is, instead of having `mNextHop `and `mRouteCost `in Router object, we could have sth. like , `mNextHop`, `mNextHopCost`, `mDirectCost`. Then we never set mNextHop to our Router ID, but we keep Router ID of a router that advertised his connection to this router (one with the smallest route cost).

In case of scenario presented by Hubert, when we lose connection to Node C, we will set `mDirectCost `to infinity, and we start using `mNextHop ` (Set to Node B). Also when we do forwarding we need to check if `mNextHopCost` is smaller than `mDirectCost `in order to use `mNextHop` field, otherwise we send frame directly.

jwhui, 2017-01-18:
Makes sense to me.  Is one of you planning to submit a PR to address this?

hubertmis, 2017-01-18:
Sorry for too short explanation. I'll create PR.

jwhui, 2017-01-26:
Fixed in #1187.

closed 2017-01-26 by jwhui