openthread/openthread #1162: Failed Cert. Test Caess  with Ti-cc2538 DK boards

bot-zjlu, 2017-01-17:
We are trying to run Cert. Test cases (not GRL) on 2 TI-CC2538 Boards by adding device management scripts. Now we got following results (for those test cases only request two boards):
1) Cert_6_1_01
2) Cert_6_3_02
3) Cert_6_4_01
4) Cert_6_5_01 
5) Cert_6_6_01
6) Cert_6_6_02(DUT as ED and SED)
7) Cert_5_3_01
8) Cert_5_5_01 (DUT as Leader, and Router)
9) Cert_5_8_02 

Failed test cases are:
 10) Cert_5_8_03
 11) Cert_8_1_01 (DUT as Commissioning and Joiner )
 12) Cert_8_1_02  
13) Cert_9_2_04  (DUT as Leader and Commissioner)
( sometimes pass with DUT as Leader)

Do you run or plan to run these test case on devices?

Question 1:  if we put any Cert_5_* test case before Cert_6_* test case,  Cert_6_* test case will fail.
                    but if after running Cert_5_* test case, and clean up the 256 bytes starting address from 0x239800, then Cent_6_* test will pass.  Do you know what the root causes?  How do you pass these test cases in your tests? 

Question 2: we know OT harness automation run GRL test cases.  But what differences between Cert. test cases and the test cases under Harness Automation (GRL)?

Question 3:  
   Some minors:
    In "Cert_6_3_02_NetworkDataUpdate.py", why is the class name "Cert_5_6_2_NetworkDataUpdate",?
    In "Cert_6_4_02_RealmLocal.py",  the class name is "Cert_5_3_2_RealmLocal"

Thanks,

jwhui, 2017-01-19:
> Question 1: if we put any Cert_5_* test case before Cert_6_* test case, Cert_6_* test case will fail. but if after running Cert_5_* test case, and clean up the 256 bytes starting address from [...]

In short, while it is in theory possible to use OpenThread's python scripts with actually hardware, we have not really used them in this way.  As a result, do not expect the python scripts to work well with actual hardware.

> Question 2: we know OT harness automation run GRL test cases. But what differences between Cert. test cases and the test cases under Harness Automation (GRL)?

The Thread Group Certification Harness is the official test harness used to test a Thread product for conformance and interoperability.  OpenThread's python scripts only implement a subset of the Thread Certification Test Plan.

> Question 3:
Some minors:
In "Cert_6_3_02_NetworkDataUpdate.py", why is the class name "Cert_5_6_2_NetworkDataUpdate",?
In "Cert_6_4_02_RealmLocal.py", the class name is "Cert_5_3_2_RealmLocal"

Those are typos.  I'll get them fixed soon.  Thanks.

bot-zjlu, 2017-01-19:
Thanks for quick response! We do need  a light weight test framewok for our
project before we get official GRL Test Harness. So it works well with some
bug fixing.

We cross check each "Cert. Test Case" against your test results of Harness
Automation, it seems to us there are more  test case in the "Cert Test
Case" than the test results listed on the web.

Thanks,

zj

closed 2017-01-25 by jwhui