openthread/openthread #1222: Problem with energy scan (test 9.2.13)

hubertmis, 2017-01-27:
Test case 9.2.13 is not robust due to CoAP retransmissions:
When node A requests node B to perform energy scan, node A sends CoAP request (CON /c/es) using node B MLEID destination address. Node B sends CoAP response (ACK /c/es) using RLOC source address. Node A cannot match ACK with the request due to address mismatch and it retransmits the request (a few times). Because of retransmissions there is a lot of unnecessary network traffic and Certification Framework has difficulties with verdict.

I'm not sure how to fix this. According to RFC 7252 CoAP client should match response with request using **address**, ports and message id (or token). But RFC 6724 does not say a word to use copy of request destination address as response source address. There are other rules to select the source address.

jwhui, 2017-01-29:
I agree the RFCs do not provide good guidance.  Regardless, is there a downside to setting the source address of the ACK to be the same as the request's destination address?  This could be done by removing the `memset` on `mSockAddr` before calling `SendEmptyAck()`.  Note that we would only want to do this if the request was unicast.  A multicast request will still need to rely on automatic source address selection.

hubertmis, 2017-01-30:
That's right. But there is also a problem with anycast addressing. I came up with idea how to fix it. I'll create a PR.

PR #1229 (2017-01-30): Use correct destination address in CoAP responses.
This PR fixes #1222 .
CoAP uses copy of request destination address as response source address. In case it is a multicast or anycast address (not allowed as source address) source selection algorithm is used.
closed 2017-01-30 by jwhui