openthread/openthread #1233: Certification test 9.2.8 fails for SED

hubertmis, 2017-01-31:
When Active Timestamp or Pending Timestamp changes a parent should send unicast Data Response a SED. The Data Response message should contain Network Data (if changed), Active Timestamp and (optionally) Pending Timestamp. If SED detects that it has outdated Timestamps it should send Data Request to the parent. Parent responses with Data Response containing Active Dataset and Pending Dataset.

In current implementation when parent detects Timestamp change it sends Data Response containing Active Dataset and Pending Dataset to child (without waiting for Data Request).

jwhui, 2017-01-31:
Thanks for pointing this out - I agree we should address this.  Do you have a fix in mind?  Otherwise, I can propose a fix.

hubertmis, 2017-01-31:
I just skimmed the code. I think that removing line 2509 (static const uint8_t responseTlvs[] = {Tlv::kNetworkData, Tlv::kActiveDataset, Tlv::kPendingDataset};) from mle_router.cpp should help. But I didn't verify this solution. I'll be out of office tomorrow and I won't be able to prepare the patch before Thursday. It would be great if you fixed it earlier, but if you have other tasks I'll do that on Thursday.

jwhui, 2017-01-31:
👍  That's what I was thinking as well.  I'll test it out and see how it goes.

librasungirl, 2017-04-06:
@hubertmis, Test Harness has been updated a lot. Do you still experience 9.2.8SED fail in newer Harness?
In fact, there is some discussion about the issue ([TESTPLAN-272](https://threadgroup.atlassian.net/browse/TESTPLAN-272), [TESTPLAN-299](https://threadgroup.atlassian.net/browse/TESTPLAN-299) in thread group). Finally it is agreed to remove the check on Data Request from SED, and unicast Data Response to SED should include Active Operational Dataset (and Pending Operational Dataset) as in current test plan.

Additionally, section 5.15.3 in Thread Specification says
> Because the Operational Datasets change much less frequently than the Network Data, the
> Operational Datasets are included in messages with the Network Data only when it is known
> that the recipient requires them.

It should be fine to include Operational Datasets in unicast Data Response to SED, what we should do might be to do some optimization - only include Operational Datasets "when it is known that the recipient requires them.", other than always include them as now.

Thoughts?

PR #1237 (2017-04-06): Remove Active and Pending Datasets from unsolicited Data Response.
Fixes #1233.
closed 2017-04-06 by jwhui