openthread/openthread #1254: Spinel: Add Source Match Radio Functionality

nibanks, 2017-02-03:
With recent changes we have added the `otLinkRaw` API and hooked most of it up to NCP and exposed it through Spinel properties. The only APIs left are related to the Source Match functionality:

```
otLinkRawSrcMatchEnable
otLinkRawSrcMatchAddShortEntry
otLinkRawSrcMatchAddExtEntry
otLinkRawSrcMatchClearShortEntry
otLinkRawSrcMatchClearExtEntry
otLinkRawSrcMatchClearShortEntries
otLinkRawSrcMatchClearExtEntries
```

I was thinking we could add 3 new properties for this functionality:

`SPINEL_PROP_MAC_SRC_MATCH_ENABLED` to support enabling/disabling, via  the `SPINEL_CMD_PROP_VALUE_SET` command.

`SPINEL_PROP_MAC_SRC_MATCH_SHORT_ADDRESSES` to support operations on the short address match list. It would support `SPINEL_CMD_PROP_VALUE_SET`, `SPINEL_CMD_PROP_VALUE_INSERT` and `SPINEL_CMD_PROP_VALUE_REMOVE`.

`SPINEL_PROP_MAC_SRC_MATCH_EXTENDED_ADDRESSES` to support operations on the extended address match list. It would function just as the short address property.

One thing to note, as there are no APIs for querying, we wouldn't be able to support `SPINEL_CMD_PROP_VALUE_GET` for any of these properties, but there is also no requirements to support a get.

Any comments?

jwhui, 2017-02-04:
@darconeous @abtink , thoughts?

abtink, 2017-02-06:
Sounds good to me.

darconeous, 2017-02-23:
Correct, there is no requirement for a GET.

Can you point me to an explanation of what exactly this source match functionality does?

If this capability has 802.15.4 PIB registers associated with it, I wonder if perhaps it might make sense to use the appropriate register from the [`PROP_15_4_PIB`](https://github.com/nibanks/openthread/blob/982ecdab9ebe05705eeafea1fd6dafcbe9ee02e1/src/ncp/spinel.h#L862) section I reserved long ago.

darconeous, 2017-02-23:
Hmm, yeah, not knowing what this functionality is, I don't see anything that looks similar in the 802.15.4 MAC PIB. :(

jwhui, 2017-02-23:
The goal of this enhancement is to export the [radio source match functionality](https://github.com/openthread/openthread/blob/master/include/platform/radio.h#L256) via Spinel.  Many IEEE 802.15.4 radios implement hardware generated ACKs and use a table of short and extended address entries to indicate when to set the Frame Pending bit in hardware generated ACKs.  While common across many radio implementations, the functionality does not directly map to IEEE 802.15.4 MAC or PHY PIBs.

PR #1371 (2017-03-01): Spinel: Radio Source Match Support
Fixes #1254. Exposes the `otLinkRaw` source match related functions to the host via new Spinel commands. Also updates otlwf.sys to use these new properties.
closed 2017-03-01 by jwhui