openthread/openthread #1257: spinel flag packing order for SPINEL_PROP_THREAD_ON_MESH_NETS

pvanhorn, 2017-02-03:
The order and format of how the flags field is unpacked in ncp_base is currently defined locally within the method; NcpBase::InsertPropertyHandler_THREAD_ON_MESH_NETS(...).

This location makes it difficult for host side implementation, as this method must be examined in order to learn the proper format.  Furthermore the maintenance of this order is challenged as the order must be defined at least in two locations. On the host side and here in the method named above.

The proposal is to define these fields in spinel.h where they can be accessed by both Host and NCP side equally.

For reference, the flags to which this issue is referring are;

    const static int kPreferenceOffset = 6;
    const static int kPreferenceMask = 3 << kPreferenceOffset;
    const static int kPreferredFlag = 1 << 5;
    const static int kSlaacFlag = 1 << 4;
    const static int kDhcpFlag = 1 << 3;
    const static int kConfigureFlag = 1 << 2;
    const static int kDefaultRouteFlag = 1 << 1;
    const static int kOnMeshFlag = 1 << 0;

jwhui, 2017-02-04:
@darconeous @abtink , thoughts?

abtink, 2017-02-06:
IIRC, the bit flags are specified by thread...
I am fine with these to be defined in either `spinel.h` or maybe even better in an openthread header file (not sure which one though...).

darconeous, 2017-03-20:
Yes, we should go ahead and specify these in the Spinel header, if they haven't been already.

darconeous, 2017-04-21:
Did we ever do this? I think we might need to continue with this.

jwhui, 2017-04-21:
This issue has not yet been addressed.

jwhui, 2017-05-16:
Resolved by #1770.

closed 2017-05-16 by jwhui