openthread/openthread #1263: Remove Nest Labs OUI from example code

aeliot, 2017-02-06:
Right now both the Posix and the Dialog example code use the Nest Labs OUI (18:B4:30).  Since these are not Nest devices, they should probably not be using our OUI. 
I'm not sure if there is a test or example range that should be used instead, more research may be needed.

aeliot, 2017-02-07:
It doesn't look like there are reserved OUI or EUI-64 ranges for development usage.

jwhui, 2017-03-10:
Any suggestions on what OUI to use?

pvanhorn, 2017-03-10:
Can/should Openthread register its own OUI?  too expensive?

turon, 2017-03-10:
Looks like public OUI is a one time fee of US $2,735.

http://standards.ieee.org/develop/regauth/oui

jwhui, 2017-03-10:
I'm not yet convinced we need to register a new OUI.  Instead, we could have a strict policy that example platforms must provide their own OUI.  The POSIX platform is really only used in emulation anyway, so using the Nest Labs (or any other organization's) OUI should not really matter.  Thoughts?

aeliot, 2017-03-11:
I'm not sure getting our own OUI is the best approach, since OpenThread isn't actually manufacturing or administrating devices.  I think an explicit policy that example platforms need to supply their own OUI is probably the best option, and probably a comment in the POSIX example where it sets the OUI.

jwhui, 2017-04-28:
Published a [Porting Guide](https://github.com/openthread/openthread/wiki/Porting-Guide) that includes the following text:

> **Note:** The `otPlatRadioGetIeeeEui64` Radio API MUST return a unique administered factory-assigned IEEE EUI-64 that includes the manufacturer's OUI.

closed 2017-04-28 by jwhui