openthread/openthread #1280: Assert in Thread::Mle::MleRouter::HandleChildIdRequest

nibanks, 2017-02-08:
I just hit the following in my testing on Windows:

```
00 ffffc681`efc8aa30 fffff808`64989823 nt!RtlAssert+0xeb [minkernel\ntos\rtl\assert.c @ 132]
01 ffffc681`efc8af90 fffff808`649800f8 otlwf!Thread::Mle::MleRouter::HandleChildIdRequest+0x963 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mle_router.cpp @ 2215]
02 ffffc681`efc8b0e0 fffff808`64980297 otlwf!Thread::Mle::Mle::HandleUdpReceive+0xb08 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mle.cpp @ 2050]
03 ffffc681`efc8b370 fffff808`6497103e otlwf!Thread::Mle::Mle::HandleUdpReceive+0x27 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mle.cpp @ 1845]
04 ffffc681`efc8b3a0 fffff808`64970ffa otlwf!Thread::Ip6::UdpSocket::HandleUdpReceive+0x2e [c:\users\nibanks\source\repos\openthread-win\src\core\net\udp6.hpp @ 147]
05 ffffc681`efc8b3d0 fffff808`64972ca4 otlwf!Thread::Ip6::Udp::HandleMessage+0x33a [c:\users\nibanks\source\repos\openthread-win\src\core\net\udp6.cpp @ 248]
06 ffffc681`efc8b450 fffff808`64972547 otlwf!Thread::Ip6::Ip6::HandlePayload+0x54 [c:\users\nibanks\source\repos\openthread-win\src\core\net\ip6.cpp @ 588]
07 ffffc681`efc8b490 fffff808`649a489a otlwf!Thread::Ip6::Ip6::HandleDatagram+0x4b7 [c:\users\nibanks\source\repos\openthread-win\src\core\net\ip6.cpp @ 761]
08 ffffc681`efc8b560 fffff808`649a558a otlwf!Thread::MeshForwarder::HandleDatagram+0x6a [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mesh_forwarder.cpp @ 2019]
09 ffffc681`efc8b5b0 fffff808`649a6065 otlwf!Thread::MeshForwarder::HandleLowpanHC+0x1fa [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mesh_forwarder.cpp @ 2005]
0a ffffc681`efc8b620 fffff808`649a616d otlwf!Thread::MeshForwarder::HandleReceivedFrame+0x2e5 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mesh_forwarder.cpp @ 1722]
0b ffffc681`efc8b6c0 fffff808`64966063 otlwf!Thread::MeshForwarder::HandleReceivedFrame+0x1d [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mesh_forwarder.cpp @ 1658]
0c ffffc681`efc8b6f0 fffff808`649676ab otlwf!Thread::Mac::Receiver::HandleReceivedFrame+0x23 [c:\users\nibanks\source\repos\openthread-win\src\core\mac\mac.hpp @ 119]
0d ffffc681`efc8b720 fffff808`6496967a otlwf!Thread::Mac::Mac::ReceiveDoneTask+0x92b [c:\users\nibanks\source\repos\openthread-win\src\core\mac\mac.cpp @ 1398]
0e ffffc681`efc8b7f0 fffff808`6495db2a otlwf!otPlatRadioReceiveDone+0xca [c:\users\nibanks\source\repos\openthread-win\src\core\mac\mac.cpp @ 1216]
0f ffffc681`efc8b830 fffff808`6494b920 otlwf!otLwfRadioReceiveFrame+0x14a [c:\users\nibanks\source\repos\openthread-win\examples\drivers\windows\otlwf\radio.c @ 472]
10 ffffc681`efc8b880 fffff801`6856b528 otlwf!otLwfEventWorkerThread+0x1010 [c:\users\nibanks\source\repos\openthread-win\examples\drivers\windows\otlwf\eventprocessing.c @ 1021]
11 ffffc681`efc8bc10 fffff801`685fec86 nt!PspSystemThreadStartup+0x48 [minkernel\ntos\ps\psexec.c @ 7675]
12 ffffc681`efc8bc60 00000000`00000000 nt!KxStartSystemThread+0x16 [d:\rs2\minkernel\ntos\ke\amd64\threadbg.asm @ 81]
```

The assert is here:

```
    switch (GetDeviceState())
    {
    case kDeviceStateDisabled:
    case kDeviceStateDetached:
        assert(false);              <============== HERE
        break;
```

My guess is that there may be an issue with detaching and still having pending packets to indicate to OpenThread. Here is a dump of the various objects on the stack:

```
class Thread::Mle::MleRouter * 0xffffe786`bc3494d8
   +0x000 mNetif           : 0xffffe786`bc348248 Thread::ThreadNetif
   +0x008 mLeaderData      : Thread::Mle::LeaderDataTlv
   +0x012 mRetrieveNewNetworkData : 
   +0x014 mDeviceState     :  ( kDeviceStateDetached )
   +0x018 mParent          : Thread::Router
   +0x044 mDeviceMode      : 0xf ''
   +0x045 isAssignLinkQuality : 
   +0x046 mAssignLinkQuality : 0 ''
   +0x047 mAssignLinkMargin : 0 ''
   +0x048 mAddr64          : Thread::Mac::ExtAddress
   +0x050 mParentRequestState :  ( kParentRequestRouter )
   +0x054 mReattachState   :  ( kReattachStop )
   +0x058 mParentRequestTimer : Thread::Timer
   +0x080 mRouterSelectionJitter : 0x1 ''
   +0x081 mRouterSelectionJitterTimeout : 0 ''
   +0x082 mLastPartitionRouterIdSequence : 0x59 'Y'
   +0x084 mLastPartitionId : 0x29f9af35
   +0x088 mChildIdRequest  : Thread::Mle::Mle::<unnamed-type-mChildIdRequest>
   +0x091 mParentRequest   : Thread::Mle::Mle::<unnamed-type-mParentRequest>
   +0x09c mParentRequestMode :  ( kMleAttachAnyPartition )
   +0x0a0 mParentLinkQuality : 0x3 ''
   +0x0a1 mParentPriority  : 0 ''
   +0x0a2 mParentLinkQuality3 : 0x1 ''
   +0x0a3 mParentLinkQuality2 : 0 ''
   +0x0a4 mParentLinkQuality1 : 0 ''
   +0x0a5 mParentLeaderData : Thread::Mle::LeaderDataTlv
   +0x0af mParentIsSingleton : 
   +0x0b0 mParentCandidate : Thread::Router
   +0x0e0 mSocket          : Thread::Ip6::UdpSocket
   +0x128 mTimeout         : 0x64
   +0x130 mSendChildUpdateRequest : Thread::Tasklet
[... 131 lines cut ...]
   +0x4a4 otPhyState       :  ( kStateReceive )
   +0x4a8 otCurrentListenChannel : 0xb ''
   +0x4a9 otReceiveMessage : [127]  "a???"
   +0x528 otTransmitMessage : [127]  "A???"
   +0x5a8 otReceiveFrame   : RadioPacket
   +0x5b8 otTransmitFrame  : RadioPacket
   +0x5c8 otLastTransmitError :  ( kThreadError_None )
   +0x5cc otLastTransmitFramePending : 0 ''
   +0x5cd otLastEnergyScanMaxRssi : 0 ''
   +0x5ce otPromiscuous    : 0 ''
   +0x5d0 otPanID          : 0xface
   +0x5d8 otFactoryAddress : 0x0a113f59`4ef01e3c
   +0x5e0 otExtendedAddress : 0xeee6c14c`6eb81267
   +0x5e8 otShortAddress   : 0xfffe
   +0x5ea otPendingMacOffloadEnabled : 0x1 ''
   +0x5eb otPendingShortAddressCount : 0 ''
   +0x5ec otPendingShortAddresses : [32] 0xc001
   +0x62c otPendingExtendedAddressCount : 0 ''
   +0x630 otPendingExtendedAddresses : [32] 0x4292265e`673c0ebc
   +0x730 otCtx            : 0xffffe786`bc308008 otInstance
   +0x738 otInstanceBuffer : 0xffffe786`bc308000  "pH???"
   +0x350 TunWorkerThread  : 0xffffe786`b1402040 Void
   +0x358 TunWorkerThreadStopEvent : _KEVENT
   +0x370 TunWorkerThreadAddressChangedEvent : _KEVENT
```

abtink, 2017-02-08:
Thanks for reporting this. 
I wonder if we may be keeping the child table when we detach... Otherwise this should have caused an early exit: 
`VerifyOrExit((child = FindChild(macAddr)) != NULL, ;);`

jwhui, 2017-02-09:
Thanks for reporting this issue! Will propose a fix soon.

PR #1290 (2017-02-09): Only handle Child ID Request when operating as child, router, or leader.
Fixes #1280.
closed 2017-02-09 by jwhui