openthread/openthread #1283: Build OpenThread code base for Host system.

pvanhorn, 2017-02-08:
A host system, one that communicates to an OpenThread NCP over a bus (UART, SPI), may have need for a modest portion of the OpenThread libraries.  These would include various header files {openthread-types.h, spinel.h, hdlc.hpp}  along with libraries that can be used to unpack spinel frames {spinel.c, hdlc.cpp}.  

The request would be to configure Openthread to build for this alternative environment.  In the future one might add additional modules that would further ease the porting/development effort of a host system.

nibanks, 2017-02-08:
Isn't this what we do for Windows already? :)

pvanhorn, 2017-02-08:
I don't know is it?  how do you configure OpenThread for Windows?

nibanks, 2017-02-08:
All the VS project files build OpenThread into a Window Driver Static Library. That is then built into the otLwf.sys Windows Filter Driver. You can find more info [here](https://github.com/openthread/openthread/tree/master/examples/drivers/windows).

pvanhorn, 2017-02-08:
Yes, it does appear that you have achieved the stated goal for windows.  I would like to achieve the same goal for other platforms as well.

nibanks, 2017-02-08:
Is your goal to completely duplicate the code we have for Windows on other platforms? What platforms specifically?

pvanhorn, 2017-02-08:
We have linux based and RTOS based platforms which act as hosts to ncp devices here at Nest.  These platforms would benefit from the ability to build the small portion of OpenThread for a host based system.  That is spinel.c, hdlc.cpp and perhaps some future ncp_host.cpp to compliment ncp_base.cpp.

closed 2018-09-12 by jwhui