openthread/openthread #1295: Division by zero in Thread::Mle::Mle::HandleChildIdResponse

vikoz, 2017-02-09:
Whilst testing, I've tried to set ROUTER_SELECTION_JITTER equal to 0. I've got the error due to division by zero:
```
00 ffffc902`0230f4d8 fffff800`d62135d2 nt!DbgBreakPointWithStatus [d:\rs2\minkernel\ntos\rtl\amd64\debugstb.asm @ 130]
01 ffffc902`0230f4e0 fffff800`d6212e7d nt!KiBugCheckDebugBreak+0x12 [minkernel\ntos\ke\bugcheck.c @ 469]
02 ffffc902`0230f540 fffff800`d618df67 nt!KeBugCheck2+0x94d [minkernel\ntos\ke\bugcheck.c @ 1756]
03 ffffc902`0230fc50 fffff800`d619abd9 nt!KeBugCheckEx+0x107 [d:\rs2\minkernel\ntos\ke\amd64\procstat.asm @ 230]
04 (Inline Function) --------`-------- nt!PspUnhandledExceptionInSystemThread+0x27 [minkernel\ntos\ps\psexec.c @ 7633]
05 ffffc902`0230fc90 fffff800`d617790f nt!PspSystemThreadStartup$filt$0+0x44 [minkernel\ntos\ps\psexec.c @ 7705]
06 ffffc902`0230fcd0 fffff800`d619442d nt!__C_specific_handler+0x9f [minkernel\crts\crtw32\misc\riscchandler.c @ 329]
07 ffffc902`0230fd40 fffff800`d60a66f2 nt!RtlpExecuteHandlerForException+0xd [d:\rs2\minkernel\ntos\rtl\amd64\xcptmisc.asm @ 135]
08 ffffc902`0230fd70 fffff800`d60a86a3 nt!RtlDispatchException+0x3f2 [minkernel\ntos\rtl\amd64\exdsptch.c @ 560]
09 ffffc902`02310460 fffff800`d6199602 nt!KiDispatchException+0x143 [minkernel\ntos\ke\amd64\exceptn.c @ 985]
0a ffffc902`02310b20 fffff800`d61962de nt!KiExceptionDispatch+0xc2 [d:\rs2\minkernel\ntos\ke\amd64\trap.asm @ 3010]
0b ffffc902`02310d00 fffff806`59e4c9c5 nt!KiDivideErrorFault+0xde [d:\rs2\minkernel\ntos\ke\amd64\trap.asm @ 264]
0c ffffc902`02310e90 fffff806`59e5043a otlwf!Thread::Mle::Mle::HandleChildIdResponse+0x745 [e:\github\openthread\src\core\thread\mle.cpp @ 2626]
0d ffffc902`023110c0 fffff806`59e505b7 otlwf!Thread::Mle::Mle::HandleUdpReceive+0xb2a [e:\github\openthread\src\core\thread\mle.cpp @ 2049]
0e ffffc902`02311350 fffff806`59e4148e otlwf!Thread::Mle::Mle::HandleUdpReceive+0x27 [e:\github\openthread\src\core\thread\mle.cpp @ 1840]
0f ffffc902`02311380 fffff806`59e4144a otlwf!Thread::Ip6::UdpSocket::HandleUdpReceive+0x2e [e:\github\openthread\src\core\net\udp6.hpp @ 147]
10 ffffc902`023113b0 fffff806`59e430f4 otlwf!Thread::Ip6::Udp::HandleMessage+0x33a [e:\github\openthread\src\core\net\udp6.cpp @ 248]
11 ffffc902`02311430 fffff806`59e42997 otlwf!Thread::Ip6::Ip6::HandlePayload+0x54 [e:\github\openthread\src\core\net\ip6.cpp @ 588]
12 ffffc902`02311470 fffff806`59e750fa otlwf!Thread::Ip6::Ip6::HandleDatagram+0x4b7 [e:\github\openthread\src\core\net\ip6.cpp @ 761]
13 ffffc902`02311540 fffff806`59e75783 otlwf!Thread::MeshForwarder::HandleDatagram+0x6a [e:\github\openthread\src\core\thread\mesh_forwarder.cpp @ 2019]
14 ffffc902`02311590 fffff806`59e76877 otlwf!Thread::MeshForwarder::HandleFragment+0x4f3 [e:\github\openthread\src\core\thread\mesh_forwarder.cpp @ 1931]
15 ffffc902`02311620 fffff806`59e769cd otlwf!Thread::MeshForwarder::HandleReceivedFrame+0x297 [e:\github\openthread\src\core\thread\mesh_forwarder.cpp @ 1714]
16 ffffc902`023116c0 fffff806`59e36293 otlwf!Thread::MeshForwarder::HandleReceivedFrame+0x1d [e:\github\openthread\src\core\thread\mesh_forwarder.cpp @ 1658]
17 ffffc902`023116f0 fffff806`59e378db otlwf!Thread::Mac::Receiver::HandleReceivedFrame+0x23 [e:\github\openthread\src\core\mac\mac.hpp @ 119]
18 ffffc902`02311720 fffff806`59e398ba otlwf!Thread::Mac::Mac::ReceiveDoneTask+0x92b [e:\github\openthread\src\core\mac\mac.cpp @ 1398]
19 ffffc902`023117f0 fffff806`59e2dcfa otlwf!otPlatRadioReceiveDone+0xca [e:\github\openthread\src\core\mac\mac.cpp @ 1216]
1a ffffc902`02311830 fffff806`59e1b960 otlwf!otLwfRadioReceiveFrame+0x14a [e:\github\openthread\examples\drivers\windows\otlwf\radio.c @ 472]
1b ffffc902`02311880 fffff800`d6086b77 otlwf!otLwfEventWorkerThread+0x1010 [e:\github\openthread\examples\drivers\windows\otlwf\eventprocessing.c @ 1021]
1c ffffc902`02311c10 fffff800`d61933d6 nt!PspSystemThreadStartup+0x47 [minkernel\ntos\ps\psexec.c @ 7704]
1d ffffc902`02311c60 00000000`00000000 nt!KxStartSystemThread+0x16 [d:\rs2\minkernel\ntos\ke\amd64\threadbg.asm @ 81]
```
in this part of the code:
```
if (mRouterSelectionJitterTimeout == 0 && numRouters < mNetif.GetMle().GetRouterUpgradeThreshold())
{
    mRouterSelectionJitterTimeout = (otPlatRandomGet() % mRouterSelectionJitter) + 1;    <== HERE
}
```
And dump of 'this' shows that `mRouterSelectionJitter : 0x0`.

jwhui, 2017-02-09:
Thanks for reporting this issue! Will propose a fix soon.

Note that this API only exists to support the Thread Certification Harness and should not be used by actual Thread products.

vikoz, 2017-02-09:
Btw, there are other places with similar code like this:
```
if ((sourceAddress.GetRloc16() == mParent.mValid.mRloc16 || router->mState == Neighbor::kStateValid) &&
            (mDeviceMode & ModeTlv::kModeFFD) &&
            (mRouterSelectionJitterTimeout == 0) &&
            (GetActiveRouterCount() < mRouterUpgradeThreshold))
        {
            mRouterSelectionJitterTimeout = (otPlatRandomGet() % mRouterSelectionJitter) + 1;
            ExitNow();
        }
```
I didn't get error there, but I reckon they're worth to look into.
Also, negative values don't seem to be handled as well.

I understand that API is only for testing purposes, but I just thought checks are always useful...
Thank you.

PR #1296 (2017-02-10): Add args check to `SetRouterSelectionJitter()`.
Fixes #1295.
closed 2017-02-10 by jwhui