openthread/openthread #1303: Disabled NCP device sends beacons after pcap mode

GiedriusM, 2017-02-10:
Title says most of it, but more details here:
Two devices, device A in ncp mode, device B in cli (though it can also be ncp).
Device A powered, wpantund launched, state is "offline". Performing scan on device B returns no networks.
Run "wpanctl pcap" for device A; performing scan on device B returns new network:
> | 0 | OpenThread       | dead00beef00cafe | ffff | 7273f4d826cb78f9 | 11 | -25 |   0 |
Cancelling pcap mode does not stop device A from sending beacons in further scans. "wpanctl status" show "offline" state.

Haven't tested if it's only the beacons, or are there any other packets.

jwhui, 2017-02-10:
Thanks for reporting this issue.  We'll investigate and get back to you.

GiedriusM, 2017-02-14:
On a somewhat related note: when should cli mode start answering to beacon requests? After **ifconfig up** or after **thread start**? I expected it to be after the latter, but it appears to be after the former.

GiedriusM, 2017-03-24:
Update: this is not NCP-only issue, but rather a MAC issue. I can see this issue while scanning during commissioning. Both, the commissioner and the joiner return beacon responses.

jwhui, 2017-03-24:
Whenever the interface is up, the IEEE 802.15.4 radio is available to communicate packets.  As a result, I believe the device should respond to IEEE 802.15.4 Beacon Requests even if Thread has not been started.

GiedriusM, 2017-03-24:
```
The beacon request command is used by a device to locate all coordinators within its radio communications range during an active scan.
```
and
```
coordinator: A device in an low rate wireless personal area network (LR WPAN) that provides synchronization services to other devices in the LR WPAN.
full-function device: A device capable of operating as a coordinator.
reduced-function device (RFD): A device that is not capable of operating as a coordinator.
```
Thread device types are very similar to these. I, personally, map them like this:
Border-routers and routers - FFD, acting as coordinators.
REED - FFD, not acting as coordinator.
SED - RFD.

Combining the definitions with the mapping, gives that only [Border] Routers should respond to beacon requests. If I'm not mistaken, REED devices in my setup do respond with beacons (can't check that ATM). However the real question is what should OpenThread do when only 802.15.4 layer is active (i.e. Thread is disabled). Could not find any specifics on this, so it may be undefined.

The way OpenThread implementation currently works and what the 802.15.4 standard says, I would conclude that each device, right from the startup (interface up) creates it's own WPAN.
IMO before Thread layer is started, the 802.15.4 layer should act as RFD. There could be otInterfaceEnableCoordinator(bool) command. By default it would be false and only when Thread device becomes a router it would be set to tru. So [border] routers would respond to beacon requests and PCAP, joiner, REED and SED "modes" - would not.

jwhui, 2017-03-24:
IEEE 802.15.4-2006 has the following definition for "coordinator":

> 3.9 coordinator: A full-function device (FFD) capable of relaying messages. If a coordinator is the principal controller of a personal area network (PAN), it is called the PAN coordinator.

At the very least, all Thread Routers should be considered IEEE 802.15.4 coordinators.  Whether or not REEDs are also considered IEEE 802.15.4 coordinators depends on the interpretation of "capable".

The primary purpose of transmitting beacons is to announce the presence of a network.  If REEDs and rx-on-when-idle end devices do not transmit beacons, then it is theoretically possible for a neighboring networks to use the same PAN ID.  Unfortunately, the way IEEE 802.15.4 ACKs work, two networks using the same PAN ID could result in ambiguous acknowledgements.

GiedriusM, 2017-03-24:
I forgot to mention, that I was referring to 802.15.4-2011 version of the standard.

Given the definition, I fully agree with you on the 3rd paragraph.

I also understand the issue with network overlapping and the need to transmit beacons on broader range of devices when the network is running. However, given the scenario when Thread stack is disabled (keeping in mind that OpenThread has no other upper 802.15.4 stacks) and that there's (at most) only one device on such intermediary "network", I fail to see a reason for devices with such states to respond to beacon requests.

jwhui, 2017-04-13:
@GiedriusM, thanks again for raising this issue.  I have submitted #1591, please review.

PR #1591 (2017-04-17): Restrict 15.4 Beacon transmissions to REEDs/Routers/Leaders.
Resolves #1303.
closed 2017-04-17 by jwhui