openthread/openthread #1304: need help to Build the command line openthread..

rmadhuraj, 2017-02-10:
HI team,
              i am trying to build the posix example to run on the Command Line , but it is throwing me error like....
       **madhu.r@CPU-245U:~/Downloads/openthread-master$ make -f examples/Makefile-posix**
.
.
.
checking for suffix of object files... configure: error: in `/home/madhu.r/Downloads/openthread-master/build/x86_64-unknown-linux-gnu':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
**make: *** [build/x86_64-unknown-linux-gnu/config.status] Error 1**

can anyone help me to solve this ..

Thanks 
Madhu R

jwhui, 2017-02-10:
Can you provide more information about your system?  What Linux distro are you using?  What version of gcc do you have installed?  Can you also attach the `config.log` file(s) here to see if those provide any clues?

rmadhuraj, 2017-02-11:
@jwhui 
            This is my system info
       **Linux CPU-245U 4.4.0-62-generic #83~14.04.1-Ubuntu SMP  x86_64  x86_64 GNU/Linux**
        
i attached a copy of config.log file by changing it as .txt file.

[config.txt](https://github.com/openthread/openthread/files/768226/config.txt)

Thanks for your time.

jwhui, 2017-02-11:
Looking at your `config.log` file, this error stands out:

> as: unrecognized option '--64'
configure:4276: $? = 1
configure:4489: checking whether we are cross compiling
configure:4527: result: yes

It appears that your toolchain has some issues.  In particular, it looks like you are mixing a newer compiler (requiring --m64 instead of just --64) with an old "as" version or mixing your normal toolchain with arm toolchain.

turon, 2017-02-11:
If you are trying to build OpenThread natively on an arm-linux system, you may need to update nlbuild-autotools to know about that toolchain:

```
# Build and install nlbuild-autotools for arm-unknown-linux
cd openthread/third_party/nlbuild-autotools/repo/tools/packages
./build
```

jwhui, 2017-02-11:
@turon, @rmadhuraj is trying to build for x86.  See his post above.

rmadhuraj, 2017-02-13:
Thanks @jwhui @turon ,
          
               so can you please tell me the possible solution?

jwhui, 2017-02-13:
@rmadhuraj , please try upgrading your GCC toolchain to something newer (e.g. 4.9, 5, or 6).  That should help resolve the `as` issue I mentioned in my previous post.

rmadhuraj, 2017-02-13:
hi @jwhui 
                i tried with 6.2 and 4.9 . still i am unable to resolve issue. i tried removing the tool chain and reinstalling the tool chain. Now i am not able to compile even simple c program. sorry , i think i am letting you to think more. 

**madhu.r@CPU-245U:~$ gcc p1.c 
as: unrecognized option '--64'**

thanks.

rmadhuraj, 2017-02-14:
HI @jwhui 
               
 i updated the gcc tool chain to gcc to gcc-6. and reinstalled the arm-gcc- tools.

also the important thing is that initially In my .bashrc file i had set many PATHS to different versions/copies of arm-gcc-none-eabi... so then i commented them and kept only one default PATH..

thanks for all the above commentS .everything helped me.
Thank you 
R Madhu.

jwhui, 2017-02-14:
@rmadhuraj, great to hear and thanks for the update.

closed 2017-02-14 by jwhui