openthread/openthread #1305: wrong panid used in discovery request.

erja-gp, 2017-02-10:
It looks like the factoryreset, resets the radio (via otPlatReset) but does not reset the panid in the openthread stack.
According to the documentation the factoryreset will “Delete all stored settings, and signal a platform reset.”
Shouldn’t the panid in the openthread stack then also not be set to 0xffff ?

This can result in the wrong panId is used during Joining.
This can be seen during e.g. during certification test 8.1.1
In my test, the DUT is controlled by the official Thread test harness (via the THCI).
The test harness performs several settings on the DUT before the test.
The settings before the joining starts are:
1) (… many other settings…)
2) factoryreset
3) commissioner provisioningurl
4) extaddr 166e0a0000000001
5) channel 20
6) dataset channel 20
7) ifconfig up
8) joiner start threadjpaketest
… at this point the discovery starts.

As you can see in the settings above, no new panId was set _AFTER_ the factoryreset (note: it is set several times to 0xface _BEFORE_ the factoryreset).

But in the sniffer log the discovery request is sent with destination panid 0xface, and no source panid. (Therefor the source panid is also 0xface)
However, given the factoryreset, the source and destination panids should have been 0xffff.

chshu, 2017-02-10:
@erja-gp Which platform are you using? We don't see this issue during Certification test with CC2538.

erja-gp, 2017-02-10:
When I look into the certification test which you mention, I see the same behaviour: please have a look here: https://github.com/openthread/openthread/files/596666/Joiner_8_1_1.zip
If you look at frame number 18, you will see that the destination panid is 0x6688, which I guess should be 0xffff. The certification test does not fail because it doesn't check the panid, but still, I believe that according to the test scenario above it should be 0xffff.

chshu, 2017-02-10:
@erja-gp, Please check the logs of Commissioner_x cases, for example: https://github.com/openthread/openthread/files/596659/Commissioner_8_1_1.zip.

In Joiner_x cases, we could control the DUT manually, which may have different setting process as you mentioned above.

jwhui, 2017-02-10:
The `factoryreset` CLI command should result in calls to `otPlatSettingsWipe()` and `otPlatReset()`.  The latter should cause the entire MCU to reset and go through `otInstanceInit()` again.  Any existing PAN ID setting should be reset to `0xffff` in the constructor.  Can you provide more information on what platform and git commit ID you are using?

erja-gp, 2017-02-13:
OK, I see. I'm not calling the constructor because I didn't restart the main function. I'll see how I can fix this.
I had created a branch 13 days ago:
https://github.com/qorvo-erja/openthread/commit/260c7e6a953eb87805c410a461921ee0421e568d
My platform consists of a Qorvo GP712 radio which is connected to a raspberry pi and I'm transferring the CLI commands from the Test Harness to the pi over a socket connection.

jwhui, 2017-03-21:
@erja-gp, is this still an issue for you?

erja-gp, 2017-03-23:
@jwhui No this no longer an issue.

closed 2017-03-23 by jwhui