openthread/openthread #1307: Windows 10 Vagrant VM

beriberikix, 2017-02-11:
For local testing, I like to use [Vagrant](https://www.vagrantup.com) to spin up different environments. I'd like to do the same for Windows tests, so I don't have to switch between physical machines. Are there free images for testing/development? I've used the [Edge VMs](https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/) in the past for browser testing but not sure about the terms of use. I also found an image for [Windows development](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines) (20gb!)

Any ideas @nibanks?

nibanks, 2017-02-11:
What exactly do you want to test with the Windows VM? Do you want the virtual Thread environment I use to run the Python cert tests?

beriberikix, 2017-02-12:
Yup, pretty much. It would also be nice to be able to run any other toolchains that are built for Windows. None do at the moment but potentially some future platforms only build and program on Windows.

nibanks, 2017-02-12:
The virtual Thread environment I use relies on the latest Windows 10 image, not yet completely released (though I believe most of it is available via the Preview images). I will see what options I have for sharing out a VM for use.

nibanks, 2017-03-03:
I hope to be sharing out a Windows 10 VHD soon that you can use, in conjunction with the [instructions on the Wiki](https://github.com/openthread/openthread/wiki/OpenThread-on%C2%A0Windows), to run though all the Python certification tests. I hope that will satisfy your requirements!

nibanks, 2017-03-09:
I have a VHD prepared and shared. I just need folks who are interested to email me (`nibanks`) `@microsoft.com` and I can give them explicit permission to download the 3.6 GB ZIP of the VHD and corresponding EULA.

beriberikix, 2017-03-09:
Thanks @nibanks! Did you find out if there's hosted (even timeboxed) VM like the ones used for Edge testing? Can those VM be used for testing OT?

nibanks, 2017-03-09:
This is the only VM that you will be able to use to run the Python certification tests with. It has some additional test drivers to expose virtual Thread interfaces.

beriberikix, 2017-03-10:
Okay! I think I can close this out. Between the option to email you for a VM and the [build instructions](https://github.com/openthread/openthread/wiki/Building-on-Windows) you already posted, everyone should be covered. Thanks again.

closed 2017-03-10 by beriberikix