openthread/openthread #1309: Spurious failures in 5.1.x tests.

jwhui, 2017-02-13:
I've been noticing an increased rate of spurious failures in the `Cert_5_1_xx.py` tests.

The tests always seem to succeed in the `state` checks, but fail on the message/sniffer check.  Example output:

```
FAIL: test (__main__.Cert_5_1_03_RouterAddressReallocation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/jwhui/openthread/examples/../tests/scripts/thread-cert/Cert_5_1_03_RouterAddressReallocation.py", line 126, in test
    leader_messages.next_mle_message(mle.CommandType.PARENT_RESPONSE)
  File "/home/travis/build/jwhui/openthread/tests/scripts/thread-cert/message.py", line 293, in next_mle_message
    assert message is not None, "Could not find MleMessage of the type: {}".format(command_type)
AssertionError: Could not find MleMessage of the type: 10
```

Some full travis logs attached:
- [cert_5_1_1.txt](https://github.com/openthread/openthread/files/770238/cert_5_1_1.txt)
- [cert_5_1_11.txt](https://github.com/openthread/openthread/files/770239/cert_5_1_11.txt)

@prfi, would you mind taking a look?

prfi, 2017-02-13:
I will check that and I will try to fix.

jwhui, 2017-02-15:
Fixed by #1324.

closed 2017-02-15 by jwhui