openthread/openthread #1316: Data Request flooding when Operational Dataset changes.

hubertmis, 2017-02-14:
When Operational Dataset changes Leader sends unsolicited Data Response to all neighbor Routers containing new Timestamp value. Every Router in response sends Data Request to get new Dataset. The problem is that all Routers send Data Request at once that causes packet collisions.
I'm not sure if MLE_MAX_RESPONSE_DELAY jitter should be used in this case to prevent packet collissions.

jwhui, 2017-02-15:
While the Thread spec does not clearly specify this behavior, I think it makes sense to apply `MLE_MAX_RESPONSE_DELAY` to MLE Data Requests sent in response to a multicast MLE Data Response.  Would you like to propose a change?

hubertmis, 2017-02-15:
Yes. I'll do that.

PR #1323 (2017-02-15): Delay transmission of Data Request in response to a multicast message.
Fixes #1316
closed 2017-02-15 by jwhui