openthread/openthread #1320: Commissioner/Joiner Router port usage is incorrect in Joiner Authentication

jovergar, 2017-02-14:
According to the the 1.1 spec, during the Joiner Authentication process, the Joiner Router should send to the Border Agent from an ephemeral port to port MM (61631), and should receive replies from the Border Agent on this ephemeral port.

joiner_router.cpp is not listening on this ephemeral port after sending the RLY_RX. Further, the Commissioner is then processing the DTLS-ClientHello and sending the RLY_TX to a hard-coded port (MM, 61631), where the Joiner Router is listening (which keeps things working in OpenThread, but is incompatible with the spec).

So there are two parts to this issue:
1) The joiner router should be listening/expecting to RLY_TX reply on whatever port it sends from rather than only on 61631. My (and I assume other 3rd party) off mesh border agent/commissioner which don't send to port 61631 end up going through the generic HandleUdpReceive path which breaks the joiner authentication.
2) The commissioner/border agent (these two roles seem to be treated the same in the codebase) should not be sending to a hard coded port 61631. A joiner router that is waiting for the RLY_TX on the port it sent from would not expect this port to be used for the reply, so OpenThread would be incompatible with third party Joiner Routers.

jwhui, 2017-02-14:
Thread 1.1 Section 8.7.1.2 indicates that the RLY_TX.ntf message from Border Agent to Joiner Router is sent to the MM port.  Based on Thread 1.1 Section 10.3.1, I believe the intent is for all CoAP Request messages to have a well-known destination port.  Can you indicate where the spec indicates that RLY_TX.ntf messages are sent to the ephemeral port?  Maybe you found an inconsistency in the spec.

Also, OpenThread does not currently support the Thread Border Agent (we are currently working on this feature).  The existing Commissioner role is the "On-Mesh Commissioner" role.

nibanks, 2017-02-14:
According to Figure 8-5 (page 223 in Thread 1.1) the RLY_TX should from the MM port, but be sent to `UDP destination port: Eph(3)`. This seems to indicate the Spec has contradicting information.

The reason Jorge encountered this is because he is adding the Windows Border Agent code.

jovergar, 2017-02-14:
I hadn't looked at 8.7.1.2 that you pointed out. Another figure that I hadn't examined too closely was 8.4.7.4.1, which agrees with the current implementation. So we have 2 sources that agree with the current implementation and that disagrees.

jwhui, 2017-02-14:
Yes, I agree that Chapter 8 is inconsistent regarding this.  Please file a SPEC JIRA so that the Thread Group's Errata Committee can work to resolve this.

jovergar, 2017-02-14:
@jwhui I'm not familiar with that process, where would I file this?

jwhui, 2017-02-14:
The Thread Group JIRA portal is at https://threadgroup.atlassian.net/

Let me know if you'd like me to file the JIRA on your behalf.

jovergar, 2017-02-15:
@jwhui I don't have access to that group, could you please file it? Thanks.

jwhui, 2017-02-15:
I have created [SPEC-701](https://threadgroup.atlassian.net/browse/SPEC-701).  Thanks for finding this spec issue!

closed 2017-02-15 by jwhui