openthread/openthread #1336: Packet dropped due to kThreadError_NoBufs when address is not in the Cache

vikoz, 2017-02-17:
During testing on Windows I've hit a weird problem with ping. I was wondering if that's my wrong understanding or if there's a bug. Could you please help me with this?
I've created a network with 27 devices (15 of them - children, other - routers).
Then I tried to ping each router from every other. Leader 0 succeeded in doing so with all routers except for the last one - 26. 
Router 26th addresses were:
```
fdde:ad00:beef:0:0:ff:fe00:4000
fdde:ad00:beef:0:6eaf:cef3:6f01:b682
fe80:0:0:0:e846:33cc:e6e1:862            (wasn't used for ping)
```
When I tried to ping the first one, I did get a response. But second one never responded.
When I tried to debug the driver, I went through the following steps:
In `ThreadError Ip6::HandleDatagram` first it determined that packet is to be forwarded:
```
}
    else                        <<<==== This means unicast, which it is
    {
        if (IsUnicastAddress(header.GetDestination()))
        {
            receive = true;
        }
        else if (!header.GetDestination().IsLinkLocal())
        {
            forward = true;   <<<===== HERE set to be forwarded
        }
```
In here it proceeded to forward the message:
```
else
{
      hopLimit = header.GetHopLimit();
      message.Write(Header::GetHopLimitOffset(), Header::GetHopLimitSize(), &hopLimit);
      SuccessOrExit(error = ForwardMessage(message, messageInfo, nextHeader));           <<=== HERE
}

```
Then `Ip6::ForwardMessage` > `ThreadNetif::SendMessage` > `MeshForwarder::SendMessage` it set the message for direct transmission and enqueued it:
```
else
        {
            // not destined for a sleepy child
            aMessage.SetDirectTransmission();
        }
...
SuccessOrExit(error = mSendQueue.Enqueue(aMessage));
```
So far everything is good.
Afterwards, in `MeshForwarder::GetDirectTransmission()` the message is retrieved:
```
switch (curMessage->GetType())
        {
        case Message::kTypeIp6:
            error = UpdateIp6Route(*curMessage);            <<==== This is being called
            break;
```
In `MeshForwarder::UpdateIp6Route`:
```
else if ((neighbor = mNetif.GetMle().GetNeighbor(ip6Header.GetDestination())) != NULL)
            {
                mMeshDest = neighbor->mValid.mRloc16;
            }
            else if (mNetif.GetNetworkDataLeader().IsOnMesh(ip6Header.GetDestination()))
            {
                SuccessOrExit(error = mNetif.GetAddressResolver().Resolve(ip6Header.GetDestination(), mMeshDest));                          <<<<===== This one is being called
            }
```
And finally in `AddressResolver::Resolve` it fails to find address in the mCache:
```
ThreadError error = kThreadError_None;
    Cache *entry = NULL;

    for (int i = 0; i < kCacheEntries; i++)
    {
        if (mCache[i].mState != Cache::kStateInvalid)
        {
            if (memcmp(&mCache[i].mTarget, &aEid, sizeof(mCache[i].mTarget)) == 0)
            {
                entry = &mCache[i];
                break;
            }
        }
        else if (entry == NULL)
        {
            entry = &mCache[i];
        }
    }

    VerifyOrExit(entry != NULL, error = kThreadError_NoBufs);              <<== HERE it returns the error
```
Going back to the `MeshForwarder::GetDirectTransmission()` the packet is being dropped due to the previous error:
```
switch (error)
        {
        case kThreadError_None:
            ExitNow();

        case kThreadError_AddressQuery:
            mSendQueue.Dequeue(*curMessage);
            mResolvingQueue.Enqueue(*curMessage);
            continue;

        case kThreadError_Drop:
        case kThreadError_NoBufs:                 <<<=== Here it's being dequeued and disposed of
            mSendQueue.Dequeue(*curMessage);
            curMessage->Free();
            continue;
```
My understanding of the sources and documentation is that after failing to find a record in the queue, node should have queried the address (and that's logical). So, `kThreadError_AddressQuery` should have been returned instead of `kThreadError_NoBufs`, and the packet should have been enqueued for Address Resolving, shouldn't it?

jwhui, 2017-02-17:
Thanks for identifying this issue.  The Address Cache should have an eviction policy when it becomes full.  I'll work on getting an LRU eviction policy implemented.

jwhui, 2017-02-21:
I submitted #1358 to address this issue.  @vikoz, can you check if #1358 resolves this issue?

PR #1358 (2017-02-22): Evict least-recently-used address cache entry.
- Maintain ordering of cache entries based on use.
- Evict least-recently-used cache entry when all are used.
- Fixes #1336.
closed 2017-02-22 by jwhui