openthread/openthread #1345: Dealing with different versions of ARM embedded gcc for examples

beriberikix, 2017-02-18:
Different examples have different required versions of ARM embedded gcc. For example, most work fine with 4.9.x but both the cc2650 and da15000 recommend 5.4.x. The standard package manager per platform (brew, apt-get, etc.) don't have a standard way to install multiple versions of the ARM gcc, making it difficult to switch between platforms. Although that might not be typical among most users, is there a best-practice or mod we can add to the convenience make files to point to different  build versions? Like a flag to override $(CC)? Use the specific named version, ie arm-none-eabi-gcc-4.9.3? 

Might also be handy for debian users as the name of the executable [changed](https://launchpad.net/gcc-arm-embedded/+announcement/13824).

jwhui, 2017-02-18:
My general process for dealing with multiple versions of tools on Linux is simply to install them into different paths.  For example, I have:

```
/opt:
    gcc-arm-none-eabi-4_7-2013q3
    gcc-arm-none-eabi-4_8-2014q3
    gcc-arm-none-eabi-4_9-2015q3
    gcc-arm-none-eabi-5_4-2016q2
    gcc-arm-none-eabi-5_4-2016q3
```

Then I simply configure my `$PATH` to point to the proper directory when compiling.

Alternatively, if you look in each of the example convenience Makefiles (e.g. [examples/Makefile-cc2538](https://github.com/openthread/openthread/blob/master/examples/Makefile-cc2538)), you'll see the following where you can specify all the tools:

```
AR                              = arm-none-eabi-ar
CCAS                            = arm-none-eabi-as
CPP                             = arm-none-eabi-cpp
CC                              = arm-none-eabi-gcc
CXX                             = arm-none-eabi-g++
LD                              = arm-none-eabi-ld
STRIP                           = arm-none-eabi-strip
NM                              = arm-none-eabi-nm
RANLIB                          = arm-none-eabi-ranlib
OBJCOPY                         = arm-none-eabi-objcopy
```

Feel free to change those to whatever executable names/paths you like.

beriberikix, 2017-02-18:
Fair enough :) I hacked together a little [shell script](https://gist.github.com/beriberikix/9d34d4d2a57baa2d360c920912190c08) because I'm lazy. Hopefully it would be useful for someone else.

closed 2017-02-18 by beriberikix