openthread/openthread #1353: Questionable CoAP behaviour in sleepy devices

GiedriusM, 2017-02-20:
Couldn't think of an informative title, so I'll try to describe the situation. I have a router and end-device(s) setup. End-device has rxOnWhenIdle set to false, therefore it receives data using MAC data request frames. From time to time this devices sends various CoAP requests (both confirmable and non-confirmable) to a server and expects a response. There are two scenarious/problems that happen:

1) With confirmable requests - device en-queues CON request message, sends it once, the request then reaches a server, server sends response, router queues this response (because of rxOnWhenIdle=false), after two seconds CoAP client's retransmit timer trigger and it sends a copy of the same request again, request reaches server, server sends response, router queues the response and this repeats for a number of configured retries. And then end-device sends MAC data request and router flushes all 5 or so identical responses to the device. IMO what should happen is that after each CoAP request send/retry the end-device should also send MAC data request (with a little delay between the two). What I mean by that you can find in https://github.com/GiedriusM/openthread/tree/hack-coap-data-req commit 16fdce0054cde90afa6dd47143875ab222cf82ed. Problem is that CoAP is separated from otInstance or any other lower layers (which is good), and "extern otInstance sInstance" had to be "hacked" into it. This is kind of dilemma - user layer can't control MAC data requests after EACH retry, but is bringing up lower layer control to CoAP a good solution? Maybe Thread Netif should check if outgoing packet is CoAP request and schedule MAC data request? What do you think is the best solution here?

2) With non-confirmable requests, that are sent multiple times per data poll period (60s). Same issue as with (1), but because non-confirmable requests are sent only once, the user layer can schedule MAC data request. However if (1) is resolved, then it could also include this one...

hubertmis, 2017-02-21:
I had the same problem but i solved it in the application code: after sending CoAP request set PollPeriod to small value like 500ms, in CoAP response handler reset PollPeriod to value used in your application.
I think this should be handled by the application. Not "hacked" in CoAP or MAC layer.

GiedriusM, 2017-02-21:
Setting poll period works and at least for the time being I'll use it, but to me it feels a lot like a workaround, rather than a solution. 
First, because it's somewhat inefficient - CoAP retries with exponential delays, and so would the number of MAC data request frames (in case no CoAP response is received). 
Second, it forces you to constantly control MAC layer from even higher layer. 
Third, this issue may not be obvious and could be not addressed in some applications resulting in higher air-time. 
Fourth, this will need to be addressed in every application (code duplication, increased bug possibility).
On a good side, it gives the flexibility to the user.
I would really like some more thoughts on this.

closed 2019-10-02 by jwhui