openthread/openthread #1354: Assert in MleRouter::HandleStateUpdateTimer

nibanks, 2017-02-20:
I was running through the Python cert tests and encountered this assert:

```
    // update children state
    for (int i = 0; i < mMaxChildrenAllowed; i++)
    {
        uint32_t timeout = 0;

        switch (mChildren[i].mState)
        {
        case Neighbor::kStateInvalid:
        case Neighbor::kStateChildIdRequest:
            continue;

        case Neighbor::kStateParentRequest:
        case Neighbor::kStateValid:
        case Neighbor::kStateRestored:
        case Neighbor::kStateChildUpdateRequest:
            timeout = Timer::SecToMsec(mChildren[i].mTimeout);
            break;

        case Neighbor::kStateLinkRequest:
            assert(false);                          <======= HERE
            break;
        }

```

Callstack:

```

00 ffff8081`64e461e0 fffff807`b01bf4b7 nt!RtlAssert+0xeb [minkernel\ntos\rtl\assert.c @ 132]
01 ffff8081`64e46740 fffff807`b01bf873 otlwf!Thread::Mle::MleRouter::HandleStateUpdateTimer+0x3a7 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mle_router.cpp @ 1808]
02 ffff8081`64e467a0 fffff807`b01a07ea otlwf!Thread::Mle::MleRouter::HandleStateUpdateTimer+0x13 [c:\users\nibanks\source\repos\openthread-win\src\core\thread\mle_router.cpp @ 1705]
03 ffff8081`64e467d0 fffff807`b019b762 otlwf!Thread::Tasklet::RunTask+0x1a [c:\users\nibanks\source\repos\openthread-win\src\core\common\tasklet.hpp @ 90]
04 ffff8081`64e46800 fffff807`b019bb49 otlwf!Thread::TimerScheduler::FireTimers+0x62 [c:\users\nibanks\source\repos\openthread-win\src\core\common\timer.cpp @ 173]
05 ffff8081`64e46840 fffff807`b017b82e otlwf!otPlatAlarmFired+0xa9 [c:\users\nibanks\source\repos\openthread-win\src\core\common\timer.cpp @ 155]
06 ffff8081`64e46880 fffff801`9d8109a7 otlwf!otLwfEventWorkerThread+0x80e [c:\users\nibanks\source\repos\openthread-win\examples\drivers\windows\otlwf\eventprocessing.c @ 867]
```

A dump of the MleRouter object and the child that asserted:

```
2: kd> ?? this
class Thread::Mle::MleRouter * 0xffffaa09`22b154d8
   +0x000 mNetif           : 0xffffaa09`22b14248 Thread::ThreadNetif
   +0x008 mLeaderData      : Thread::Mle::LeaderDataTlv
   +0x012 mRetrieveNewNetworkData : 
   +0x014 mDeviceState     :  ( kDeviceStateLeader )
   +0x018 mParent          : Thread::Router
   +0x044 mDeviceMode      : 0xf ''
   +0x045 isAssignLinkQuality : 
   +0x046 mAssignLinkQuality : 0 ''
   +0x047 mAssignLinkMargin : 0 ''
   +0x048 mAddr64          : Thread::Mac::ExtAddress
   +0x050 mParentRequestState :  ( kParentIdle )
   +0x054 mReattachState   :  ( kReattachStart )
   +0x058 mParentRequestTimer : Thread::Timer
   +0x080 mDelayedResponseTimer : Thread::Timer
   +0x0a8 mRouterSelectionJitter : 0x78 'x'
   +0x0a9 mRouterSelectionJitterTimeout : 0 ''
   +0x0aa mLastPartitionRouterIdSequence : 0 ''
   +0x0ac mLastPartitionId : 0
   +0x0b0 mDelayedResponses : Thread::MessageQueue
   +0x0b8 mChildIdRequest  : Thread::Mle::Mle::<unnamed-type-mChildIdRequest>
   +0x0c1 mParentRequest   : Thread::Mle::Mle::<unnamed-type-mParentRequest>
   +0x0cc mParentRequestMode :  ( kMleAttachAnyPartition )
   +0x0d0 mParentLinkQuality : 0 ''
   +0x0d1 mParentPriority  : 0 ''
   +0x0d2 mParentLinkQuality3 : 0 ''
   +0x0d3 mParentLinkQuality2 : 0 ''
   +0x0d4 mParentLinkQuality1 : 0 ''
   +0x0d5 mKeepAliveAttemptsSent : 0 ''
   +0x0d6 mParentLeaderData : Thread::Mle::LeaderDataTlv
   +0x0e0 mParentIsSingleton : 
   +0x0e4 mParentCandidate : Thread::Router
   +0x110 mSocket          : Thread::Ip6::UdpSocket
   +0x158 mTimeout         : 0x64
   +0x160 mSendChildUpdateRequest : Thread::Tasklet
   +0x180 mDiscoverHandler : (null) 
   +0x188 mDiscoverContext : (null) 
   +0x190 mIsDiscoverInProgress : 
   +0x191 mAnnounceChannel : 0xb ''
   +0x192 mPreviousChannel : 0 ''
   +0x194 mPreviousPanId   : 0xffff
   +0x198 mLeaderAloc      : Thread::Ip6::NetifUnicastAddress
   +0x1c0 mLinkLocal64     : Thread::Ip6::NetifUnicastAddress
   +0x1e8 mMeshLocal64     : Thread::Ip6::NetifUnicastAddress
   +0x210 mMeshLocal16     : Thread::Ip6::NetifUnicastAddress
   +0x238 mLinkLocalAllThreadNodes : Thread::Ip6::NetifMulticastAddress
   +0x250 mRealmLocalAllThreadNodes : Thread::Ip6::NetifMulticastAddress
   +0x268 mNetifCallback   : Thread::Ip6::NetifCallback
   +0x280 mAdvertiseTimer  : Thread::TrickleTimer
   +0x2d8 mStateUpdateTimer : Thread::Timer
   +0x300 mChildUpdateRequestTimer : Thread::Timer
   +0x328 mAddressSolicit  : Thread::Coap::Resource
   +0x348 mAddressRelease  : Thread::Coap::Resource
   +0x368 mRouterIdSequence : 0x7 ''
   +0x36c mRouterIdSequenceLastUpdated : 0x3c11f
   +0x370 mRouters         : [63] Thread::Router
   +0xe44 mMaxChildrenAllowed : 0x20 ' '
   +0xe48 mChildren        : [32] Thread::Child
   +0x2348 mChallengeTimeout : 0x1 ''
   +0x2349 mChallenge       : [8]  "U???"
   +0x2352 mNextChildId     : 0x1ff
   +0x2354 mNetworkIdTimeout : 0x78 'x'
   +0x2355 mRouterUpgradeThreshold : 0x10 ''
   +0x2356 mRouterDowngradeThreshold : 0x17 ''
   +0x2357 mLeaderWeight    : 0x40 '@'
   +0x2358 mFixedLeaderPartitionId : 0
   +0x235c mRouterRoleEnabled : 
   +0x235d mRouterId        : 0x34 '4'
   +0x235e mPreviousRouterId : 0x34 '4'
   +0x2360 mPreviousPartitionId : 0x275f3cd6
2: kd> ?? this->mChildren[0]
class Thread::Child
   +0x000 mMacAddr         : Thread::Mac::ExtAddress
   +0x008 mLastHeard       : 0x3bd24
   +0x00c mValid           : Thread::Neighbor::<unnamed-tag>::<unnamed-type-mValid>
   +0x00c mPending         : Thread::Neighbor::<unnamed-tag>::<unnamed-type-mPending>
   +0x018 mKeySequence     : 0
   +0x01c mState           : 0y100 ( , kStateLinkRequest )
   +0x020 mMode            : 0y1111
   +0x020 mDataRequest     : 0y0
   +0x021 mLinkFailures    : 0 ''
   +0x022 mLinkInfo        : Thread::LinkQualityInfo
   +0x028 mIp6Address      : [6] Thread::Ip6::Address
   +0x088 mTimeout         : 0x64
   +0x08c mFragmentOffset  : 0
   +0x090 mIndirectSendMessage : (null) 
   +0x098 mRequestTlvs     : [5]  ""
   +0x098 mAttachChallenge : [8]  ""
   +0x0a0 mNetworkDataVersion : 0 ''
   +0x0a2 mQueuedIndirectMessageCnt : 0
   +0x0a4 mAddSrcMatchEntryShort : 0y0
   +0x0a4 mAddSrcMatchEntryPending : 0y0
```

jwhui, 2017-02-21:
Thanks for reporting this issue.  I'll take a look.

jwhui, 2017-02-21:
Just submitted #1355.  Can you check if that resolves this issue?  Thanks.

PR #1355 (2017-02-22): Do not setup neighbor state when responding to MLE Link Request from child.
- Fixes #1354.
closed 2017-02-22 by jwhui