openthread/openthread #1366: Move updated wiki sitenav

beriberikix, 2017-02-22:
@betsyl-at-google has a draft of a [new sitenav](https://github.com/openthread/openthread/wiki/OpenThread-Docs) for the wiki that's more useful than we have now. Should we integrate it into the wiki home page?

beriberikix, 2017-02-22:
On second thought, and probably the more obvious one, we could replace the existing sidenav.

betsyl-at-google, 2017-02-23:
Let me know if you want my help with this.

beriberikix, 2017-03-07:
@betsyl-at-google @jwhui can you review the proposed nav I just staged at the wiki?

nibanks, 2017-03-07:
Would it be ok to add a link to [Building on Windows](https://github.com/openthread/openthread/wiki/Building-on-Windows) to either the [Getting Started](https://github.com/openthread/openthread/wiki/OpenThread-Docs#get-started) or [Build Tools](https://github.com/openthread/openthread/wiki/OpenThread-Docs#build-tools) sections? Also, I'm not sure where we should link in the [OpenThread on Windows](https://github.com/openthread/openthread/wiki/OpenThread-on%C2%A0Windows) installation instructions wiki page.

beriberikix, 2017-03-07:
How about Building on Windows under Getting Started and OpenThread on Windows in the Platforms page?

nibanks, 2017-03-07:
Works for me. Thanks.

closed 2017-03-10 by beriberikix