openthread/openthread #1377: Question about Joiners management in Commissioner

rlubos, 2017-02-23:
I have a question regarding Commissioner and Joiners management. Current implementation seems to only support manual Joiner removal. It seems quite inconvinient to remove it manually anytime Joiner attaches, otherwise the buffer for Joiners gets filled up easily.

I would like to introduce some automation to the Joiner removal process, and I see two options to implement it:
1. One option is to implement it directly in Commissioner class and remove Joiner after succesfully sending Joiner Finalize Response. As I checked the code, it shall not impact the possible retransmissions of this message. Additionally, a timer could be added to remove Joiners after some time, in case commissioning failed / did not happen at all.
2. Second option is to provide some callback funtion to the application, and call it whenever Commissioning process ends. The application would be responsible for removing Joiners and possible timeout handling.

Personally I would opt for option 1, as it seems neat and would not require to modify for example a NCP code, but I would like to see your opinion.

hubertmis, 2017-02-24:
I like option 1. I would consider adding OpenThread API to enable/disable automatic Joiner removal.

nibanks, 2017-02-24:
I would also vote for option 1.

jwhui, 2017-02-27:
Thanks for proposing this enhancement!  I also prefer Option 1.  I think we should make the timeout mandatory (users can always add a long timeout if they wish).  I am also fine with removing the Joiner after sending the Joiner Finalize Response.  However, if/when OpenThread supports application-layer provisioning through the DTLS session, we may need to add hooks so that the application can indicate when application-layer provisioning has completed successfully.

PR #1421 (2017-03-08): Automatic Joiners removal in Commissioner
This PR resolves #1377.

Commissioner will now remove Joiners after successfull commissioning or after a timeout period.

Again, I've hit the RAM shortage on CC2538, hence I did some memory savings in JoinerRouter class. 8 bytes only, but it was enough.
closed 2017-03-08 by jwhui