openthread/openthread #1394: Endless MAC Poll Timer

nibanks, 2017-02-27:
I'm not exactly sure how I got into this state, but I think it happened when I `thread stop` and `ifconfig down` on an `otInstance`; possibly while there was an outstanding Transmit. My Windows device got into a tight loop endlessly starting a 0 ms timer:

```
09:46:46.702    [otCore]    ---> otPlatAlarmFired
09:46:46.702     [otLwf]    otPlatAlarmStop+otPlatAlarmStop
09:46:46.702    [otCore]    MAC Poll timer fired with DataRequest in SendQueue.
09:46:46.702     [otLwf]    otPlatAlarmStartAt 0 ms
09:46:46.702    [otCore]    <--- otPlatAlarmFired
09:46:46.702    [otCore]    ---> otPlatAlarmFired
09:46:46.702     [otLwf]    otPlatAlarmStop
09:46:46.702    [otCore]    MAC Poll timer fired with DataRequest in SendQueue.
09:46:46.702     [otLwf]    otPlatAlarmStartAt 0 ms
09:46:46.702    [otCore]    <--- otPlatAlarmFired
```

I don't have any other logs than these. Just 100 MB of those logs, collected in a matter of a minute. It rendered the machine completely unresponsive.

nibanks, 2017-02-27:
I have confirmed that `thread stop` is what puts the machine in this state.

pvanhorn, 2017-02-27:
The poll timer will certainly reset when there is already a datarequest in the sendqueue.  Its a bit of a mystery as to why the mPollPeriod == 0?  I wonder if as part of calling Thread stop or ifconfig down, the MeshForwarder class re-initialized via its constructor (mPollTimer is initialized to 0 there)? The logic in MeshForwarder::Stop looks like it should solve this issue as it calls mPollTimer.Stop() and dequeues all messages in the sendQueue.  Can you confirm the MeshForwarder::Stop is getting called in your test case? After that call no more frames should be submitted to mSendQueue.

pvanhorn, 2017-02-27:
Perhaps you have an order of operations issue.  It looks like if your code executed the MeshForwarder constructor before calling MeshForwarder.Stop() you could find yourself in the situation that you describe.  MeshForwarder.Stop() is called by Thread::Down().  I wonder if thread stop is calling the constructor?

nibanks, 2017-02-27:
I put a break point on `otlwf!Thread::MeshForwarder::Stop` and it was not hit when I executed `thread stop`. My repro is to run the following with only one device:

```
panid 0xface
ifconfig up
thread start
// wait a few seconds
state // -> leader
thread stop
```

And when `thread stop` is executed there is a pending transmit, if that makes any difference.

pvanhorn, 2017-02-27:
Should the proper order be ifconfig down followed by thread stop rather than the other way around?

nibanks, 2017-02-27:
I don't think calling either order should cause this kind of behavior. But I still believe `thread stop` should be called before `ifconfig down` anyways.

LuDuda, 2017-02-27:
I believe this is the path of the problem:

`otThreadStop` -> `Mle::Stop` -> `Mle::SetStateDetached` -> `MeshForwarder::SetRxOnWhenIdle(false)` -> `mPollTimer.Start(mPollPeriod)`

PollTimer is started with uninitialized mPollPeriod of 0.

I think this has been introduced here: https://github.com/openthread/openthread/commit/63b78808c356ea86904991dbf9f4b590f81e0593

The previous version called SetRxOnWhenIdle with parameter `true` instead of `false`.

pvanhorn, 2017-02-28:
I agree with @LuDuda that code alone is always going to be a problem.

jwhui, 2017-02-28:
@nibanks, thanks for reporting this issue!

@LuDuda, good catch!  Do you already have a fix?  If not, I can propose one soon.

LuDuda, 2017-02-28:
Yes i will try to propose an easy fix.

PR #1403 (2017-03-01): Fix polling behaviour on Thread Stop.
This PR fixes #1394
closed 2017-03-01 by jwhui