openthread/openthread #1398: cc2650 and sniffer.py

beriberikix, 2017-02-28:
What is required to use the cc2650 with the [sniffer.py](https://github.com/openthread/openthread/blob/master/tools/spinel-cli/sniffer.py) tool? I flashed the latest NCP bin (thanks again @srickardti @chshu!) and plugged in a usb cable. However, since the port is connected to the JTAG debugger and not your typical usb2serial, I don't know how to tell `sniffer.py`which connection to use. Can the XDS110 be configured to act as a uart? Do I need an external usb2serial device and if so, which pins should be used? Am I missing something obvious?

beriberikix, 2017-02-28:
Nevermind, user error! The device shows up correctly as a usbmodem and I was able to pipe data to wireshark.

closed 2017-02-28 by beriberikix

turon, 2017-02-28:
Glad you were able to figure it out!  

Just for reference, the various connection methods are supplied in `tools/spinel-cli/spinel/stream.py`.  Currently supported connection methods are uart, socket, and pipe.  More methods could be added if they are determined to be useful.

```
$ ./sniffer.py --help
Usage: sniffer.py [options]

Options:
  -h, --help            show this help message and exit
  -u UART, --uart=UART  
  -p PIPE, --pipe=PIPE  
  -s SOCKET, --socket=SOCKET
  -n NODEID, --nodeid=NODEID
  -q, --quiet           
  -v, --verbose         
  -d DEBUG, --debug=DEBUG
  -x, --hex             
  -c CHANNEL, --channel=CHANNEL
```

jwhui, 2017-02-28:
I believe the issue was more about how the CC2650 presents its serial interface, rather than how to use `sniffer.py`.