openthread/openthread #1405: Assert in network_data_leader_ftd.cpp

LuDuda, 2017-03-01:
I step into assert inside `network_data_leader_ftd.cpp` 

`network_data_leader_ftd.cpp:818: ThreadError Thread::NetworkData::Leader::RemoveRloc(uint16_t): Assertion false failed. Aborted (core dumped)`

This happens when Border Router that is already a native commissioner tries to register a prefix. I am working on that right now, i see that we have probably too many asserts in this file, i am trying to replace them with simple error return, or do you have other ideas on how to handle it?

jwhui, 2017-03-01:
I agree we should remove the asserts that check for unknown TLVs.  Those are left over from initial implementation and validation.  For backwards compatibility, the implementation needs to be able to handle unknown TLVs in case new ones are defined in the future.

pvanhorn, 2017-03-01:
Asserts during project development and debug are usually a good way to quickly catch and fix bugs.  Asserts in release code however are bad news if the error in question can be survived.  Perhaps what we need is a configurable macro that can take on the behavior of an assert during development and that of a simple error code return inside production code?

LuDuda, 2017-03-01:
@jwhui @pvanhorn Thanks for feedback. I have in my local branch version that replaces assert with `kThreadError_Parse ` error but i  see for backwards compatibility we should probably just remove them then, right?

jwhui, 2017-03-01:
Yes, I think it makes sense to simply skip over TLVs when possible.

I think a good general rule of thumb for TLV processing is to "be liberal in what you accept".  Of course, there are cases where the Thread spec requires more strict rules, but where those cases are not specified, we should be more liberal.  I believe this should lead to an overall more robust networking solution, especially when you cannot control other endpoints in the system.

PR #1406 (2017-03-01): Remove asserts when unknown TLV type is found
This PR fixes #1405 by ignoring unknown TLVs rather than asserting.

No parse errors are return in place of asserts to ensure backwards compatibility with future versions of Thread protocol.
closed 2017-03-01 by jwhui