openthread/openthread #1407: Platform Tx Power API

pvanhorn, 2017-03-02:
There currently does not exist a mechanism by which Openthread can initialize the TxPower used by the platform for frames other than those that OpenThread transmits explicitly.  Consider ACK frames which are typically generated by either the hardware or radio driver in response to received frames.  The TX power used for these auto-generated frames must be configured.  

It is proposed that the otSetMaxTransmitPower API be extended to also call a new API otPlatRadioSetDefaultTxPower.  This call would be in addition to the existing behavior of otSetMaxTransmitPower which should remain unchanged.

jwhui, 2017-03-02:
I agree that OpenThread needs to support configuring the "default" transmit power at the radio level.  Your proposal seems reasonable to me.

pvanhorn, 2017-03-03:
@jwhui This can be closed.

jwhui, 2017-03-03:
Fixed by #1417.

closed 2017-03-10 by jwhui