openthread/openthread #1408: Drop of packet due to invalidated neighbours

vikoz, 2017-03-02:
When I run some test on Windows, my ping function fails. Usually after number of nodes is ~20 or more. They are all in router states, and they're neighbours of one another.
After debugging a bit, it appeared that the reason of failure lies in `Mac::ReceiveDoneTask` here:
```
if (neighbor == NULL)
{
    otLogDebgMac("drop not neighbor");                         <<===== HERE
    ExitNow(error = kThreadError_UnknownNeighbor);
}
```
When I looked into why neighbor is NULL, I found out that `this->mNetif->mMleRouter->mRouters` has all of its records (aka neighbor routers) in an invalid state, including the sender of a packet.
It seems to me like the neighbour timeout has expired and they were invalidated. But I don't know why weren't they queried again or something.
Also, when I query neighbour using the node API, it returns that they are neighbours.
And unfortunately as for now this is all I've discovered about this issue, because this 'invalidation' happens to a random node, and before that happens it's not clear which one will fail, hence catch the moment when the neighbours are invalidated is very hard.

I was wondering if you would probably know what might be the issue, or if you could give me some pointers on where to look and what to check?

jwhui, 2017-03-02:
Thanks for reporting this issue.  Do you have OpenThread logging enabled?  If so, can you provide any log output?

vikoz, 2017-03-02:
I do have some logs, though they seem not quite complete... Anyway, I have attached output from my test along with the OpenThread logs.
[logs.zip](https://github.com/openthread/openthread/files/814875/logs.zip)

vikoz, 2017-03-02:
I do have some logs, although they seem to be not complete. Anyway, I have attached output from my test along with OpenThread driver's logs.
[logs.zip](https://github.com/openthread/openthread/files/814993/logs.zip)

jwhui, 2017-03-02:
Thanks for providing the logs.  I looked through them, however I am not able to make much sense of them.

Some observations:
- All "drop not neighbor" log entries occur around 09:42:15.
- Coincidentally, all "delete router id" log entires also occur around 09:42:15 on thread 05F8.
- There is a big time gap in the log:

```
3/1/17 17:47:38.225     0004    1ADC                      [otCore]      ---> otPlatRadioReceiveDone kThreadError_None
3/1/17 17:47:48.479     0004    1ADC                      [otCore]      MAC Received from short address 5c00
3/2/17 09:40:45.386     0004    1ADC                      [otCore]      MAC drop not neighbor
3/2/17 09:40:55.416     0004    1ADC                      [otCore]      MAC Dropping received frame, kThreadError_UnknownNeighbor
3/2/17 09:41:16.177     0004    1ADC                      [otCore]      <--- otPlatRadioReceiveDone
3/2/17 09:42:14.955     0004    1ADC                       [otLwf]      <--- otLwfRadioReceiveFrame
```

Can you help provide some background on the log and test scenario?

vikoz, 2017-03-03:
Yes, neither could I make much sense of them. I am currently working on making them clearer, and then parsing them and separating for different nodes. Thing is that there are multiple nodes running on one machine and logs are mixed for all of them. Yes, they could be distinguished by thread id, but this is a complicated way. So, I will provide you with another log file shortly. But it takes long time for me because after merging the latest changes from master this issue seems to have become much harder to reproduce...

jwhui, 2017-03-29:
@vikoz, are you still running into this issue?

jwhui, 2017-04-13:
@vikoz, is this still an issue for you?  If not, we should close this out.

jwhui, 2017-04-21:
Closing this out since it's gone stale.  Please reopen if this is still an issue.

closed 2017-04-21 by jwhui