openthread/openthread #1410: Out-of-tree platform implementation

wbober, 2017-03-02:
The current source code organization doesn't allow for out-of-tree implementation of a platform. This is because flash.h is not part of the public API and settings.cpp are not included in the library.

Is there any strong reason to hide them? If not then I propose to:
a) move flash.h from examples/platforms/utils to include/platform (changing prefix name from util* to ot*) b) move settings.cpp from  examples/platforms/utils to src/core/utils (?)

jwhui, 2017-03-02:
We originally had `otPlatFlash*` APIs as you are suggesting here.  OpenThread switched to the current model with #889.  The argument was that use of `otPlatFlash*` APIs were implementation and platform specific, whereas OpenThread platform APIs are intended to be implementation and platform agnostic.  In this case, the `settings.cpp` file was an implementation of `otPlatSettings*` that flash-based systems may choose to use.  However, `settings.cpp` would not apply to non-flash-based systems and flash-based systems may choose to implement `otPlatSettings*` in a different way.

With that background, happy to revisit and discuss this decision again.  @wbober, @pvanhorn, and any others please share any additional thoughts you may have.

wbober, 2017-03-02:
@jwhui Thanks. Ok, I understand the motivation. Maybe we could make the flash.h API public but put it, for example, to `include/utils` rather than to `include\platform`? This way it would be still clear what API must be implemented for a platform but one could use the settings implementation if he wanted to.

jwhui, 2017-03-10:
@pvanhorn, as author of #889, can you provide input here?

pvanhorn, 2017-03-10:
As long as flash.h remains somewhere under examples, I have no objection.  If you are suggesting to move flash.h back into the core of openthread, then I have some concerns as that is not an API that we intend to have Openthread ever use directly.  Openthread loses interest after the otSettings API. Truth be told there is no expectation that setting should even reside in flash.  It may be stored in another resource and that would be equally acceptable.

If a developer wishes to adopt flash.h for their own use, they are free to do so, however Openthread has no desire to support that API as it would other code which resides in its core directories.

wbober, 2017-03-13:
@pvanhorn 
That only works if we split configuration options for utils (`examples\platform\utils`) from platforms and apps (`examples\platforms\* and `examples\apps`).  Would that work for you?

pvanhorn, 2017-03-13:
@wbober 
I think that would work for me.  I would need to see the final product to be sure, but it sounds as if you are merely moving things around within the examples sub-directory in a way that would make your goals achievable and not break any existing features in the process.  If that assessment is accurate, then I would be in favor of this change.

jwhui, 2017-04-03:
@wbober, any updates on this issue?

wbober, 2017-04-04:
@jwhui I'll work in it in next few days.

wbober, 2017-04-05:
I've submitted a PR to this. However, this is not exactly what requested as `flash.h` gets exported to `openthread\platform\utils`. I'm not sure how to solve it otherwise, since this file is needed so I can build platform out-of-tree.

I'm not sure if autoconf can be configured to place this file in `openthread\platform\examples`, but assuming it can, would that be a ok for you?

@pvanhorn @jwhui What do you think?

wbober, 2017-04-05:
I've submitted a PR to fix this. I got it wrong with `flash.h` which is not really needed for an out-of-tree build.

jwhui, 2017-04-06:
Resolved by #1557.

closed 2017-04-06 by jwhui