openthread/openthread #1425: Build: Compiling openthread should not cause build machine to reboot.

turon, 2017-03-03:
I have a fairly stock Ubuntu 16.04 virtual machine that would reboot randomly whenever I would initiate the the stock make commands to build an OpenThread example platform, i.e. the `make -f examples/Makefile-posix` method.

I've isolated the root cause to the large default values for the number of build jobs (`BuildJobs ?= 99`).  I plan to submit a PR to resolve this.

We probably want the default value to be conservatively low so new developers don't have a bad out-of-box experience, and let Travis dial up the numbers within the Continuous Integration scripts.

jwhui, 2017-03-03:
Are others experiencing this problem?

jwhui, 2017-03-03:
FWIW, Travis uses Ubuntu Trusty VMs (which should be 16.04).

wbober, 2017-03-04:
@jwhui @turon  I use Arch based VM exclusively. I don't have any issues with building for NRF52840 and I don't recall to have any with Posix builds (although I haven't tried for some time).

Gr3yh0und, 2017-03-05:
Haven't seen this on Debian Stretch yet ...

commit a17c6876 (2017-03-08): Fix #1425: Reduce default BuildJobs to prevent reboot during build. (#1426)

* Fix #1425: Reduce default BuildJobs to prevent random reboots during build.

* Always use default BuildJobs value in .travis/script.sh.
closed 2017-03-08 by jwhui