openthread/openthread #1435: settings: handling compatibility between different versions

jwhui, 2017-03-07:
As raised by @nibanks in #1423, we need to establish some method for handling different versions of settings stored using the `otPlatSettings*` API.  This issue is solicit thoughts around what method to use.

As a starting point, it seems like backward compatibility is required to allow migration from an older version of OpenThread to a newer version.  However, is forward compatibility also required - to allow migration from a newer version of OpenThread back to an older version?

If both forward and backward compatibility is required, I'd argue that in most cases, settings structures are only allowed to add fields by appending to the end.  Fields should generally not be removed.

pvanhorn, 2017-03-07:
We can never be sure what the life cycle of a product will require, but I think its fair to imagine that some product will have a need for forward compatibility. Perhaps it needed to revert a distribution and that might entail forward compatibility support.  I think your rule about never removing structures is spot on. If we are storing sequence numbers we may also need to store those in multiple locations for the sake of forwards compatibility.  Also, we should ensure that the NVM code can safely traverse any records that it doesn't recognize.

nibanks, 2017-03-07:
While my personal preference is to use explicit version numbers, but I am fine if we decide to use the model of "only add properties" and update the code to account for a variable being missing, and set some default. 

But if we do encounter a scenario where we have to delete, then our only recourse will be to start using a new ID, right? How do we go about migrating the data from the previous ID to the new one?

jwhui, 2017-04-10:
After thinking about this issue for some time, here is my proposal:

1. Modifying existing [otPlatSettings*](https://github.com/openthread/openthread/blob/master/include/openthread/platform/settings.h)  key values MUST only be done by appending new fields.  Existing fields MUST NOT be deleted or modified in any way.
2. To support backward compatibility (rolling back to an older software version), code reading and processing key values MUST process key values that have longer length.  Additionally, newer versions MUST update/maintain values in existing key value fields.
3. To support forward compatibility (rolling forward to a newer software version), code reading and processing key values MUST process key values that have shorter length.
4. New Key IDs may be defined in the future with the understanding that such key values are not backward compatible.

Once we stabilize, the persistent settings required by Thread should be very infrequent.  Given that, I don't think we need to have a more involved versioning mechanism.

Thoughts?

nibanks, 2017-04-10:
I'm fine with this design. I would also recommend someone doing a pass on all existing settings to make sure they are in an acceptable state going forward. Anything unnecessary should be removed now, so we have no need to remove anything in the future.

chshu, 2017-04-11:
@jwhui, I totally agree with the proposal, and I also agree with @nibanks that we have to refine the settings to make sure we don't need to remove any fields in the future.

After going through all existing settings, I think **ChildInfo** and **ParentInfo** needs to be refined. I will reopen #1572 which has already addressed ChildInfo, I will also refine ParentInfo and push the PR soon.

Please let me know if you have any other suggestions, thanks.

PR #1754 (2017-05-11): Co-locate settings structs and document rules for updating.
Resolves #1435.
closed 2017-05-11 by jwhui