openthread/openthread #1444: Test case 9.2.12 problem

hubertmis, 2017-03-09:
When nodes listening on a channel receive MLE Announce with greater Active Timestamp and other channel, they send simultaneously Parent Request message on the other channel. It causes packet collisions. Perhaps there should be jitter when sending Parent Request in response to the Announce message.

LukaszGolonkaTieto, 2017-03-10:
I am able to reproduce that issue.

jwhui, 2017-03-10:
Thanks for reporting this issue!  This is yet another place where a message is sent in response to receiving a multicast message.  In general, I believe we should be applying jitter to all messages sent in response to receiving a multicast message.  @hubertmis, would you like to propose a fix?

hubertmis, 2017-03-13:
I'm out of office this week. I could have a look at this problem next week.

hubertmis, 2017-03-17:
I've started working on this issue. I found another problem. I'll explain it using simple example:
There are 3 nodes in the network: A, B, and C. A is in link rage of B, B is in link range of A and C, C is in link range of B. A is using channel 11 and has ActiveTimestamp == 30 while B and C are using channel 12 and have ActiveTimestamp == 10. A sends Announce message on channel 12. B receives Announce message and attaches to A on channel 11. After successful attachment B sends Announce message on channel 12 3 times (1 second period). When C receives Announce message it sends Parent Request on channel 11 but there is no response (A is out of range and B is using channel 12 after sending MLE Announce on channel 12). It seems that B should enable it's receiver on channel 11 after sending Announce message on channel 12.
I think the best way to fix it is to update Radio platform documentation and implementation to switch to receive channel after successful transmission. We can also add `otPlatRadioReceive` call with receive channel in `otPlatRadioTransmitDone` handler.

jwhui, 2017-03-17:
A call to `otPlatRadioTransmitDone()` should normally result in the following call path:

1. `Mac::TransmitDoneTask()`
2. `Mac::SentFrame()`
3. `Mac::ScheduleNextTransmission()`
4. `Mac::NextOperation()`
5. `otPlatRadioReceive()`

On return from this path, the receive channel should be set back to the `Mac::mChannel` value.

That said, one issue I do see is that on a Channel Access Failure, Abort, or No Ack error, `Mac::SentFrame()` will not call `Mac::ScheduleNextTransmission()` (and thus never call `otPlatRadioReceive()`.  Do you think this is the issue you are running into?

hubertmis, 2017-03-20:
You are right. I didn't follow this path deep enough. I came across this issue on Posix platform, so there is probably race in Posix radio implementatin. I'll try to fix that.

jwhui, 2017-03-21:
FWIW, I am seeing increased spurious failures on 9.2.12 in Travis.

PR #1494 (2017-03-22): Test 9.2.12 fixes
This PR fixes #1444 .

It adds jitter before sending Parent Request message to avoid packet collisions after receiving Announce message.

This change showed problem with attachment procedure just after parent candidate becomes Router: before candidate established route path to the Leader it did not respond to Parent Request messages. I added initial path to the Leader (when upgrading to Router state) to use previous parent as mNextHop and previous parent's cost as mCost. Using this data, parent candidate can send Parent Response. The node that upgraded to the Router state upd [... cut]
closed 2017-03-22 by jwhui