openthread/openthread #1446: OpenThread deinitialization

hubertmis, 2017-03-09:
When OpenThread is initialized it takes some peripherals. It is not possible to deinitialize the OpenThread stack and use taken peripherals by other modules. It seems to be a good idea to provide termination function that releases OpenThread resources if an application don't use Thread connectivity continuously.

jwhui, 2017-03-09:
We have an [`otInstanceFinalize()`](https://github.com/openthread/openthread/blob/master/include/openthread/instance.h#L197) function that is intended to serve this purposes.  Maybe that implementation needs to be extended to support your use cases?

hubertmis, 2017-03-10:
I meant something like `PlatformDeinit()`. Maybe I'll just implement it and create a PR. We'll discuss there.

closed 2017-03-13 by hubertmis