openthread/openthread #1459: Responses to MLE requests sent by SED

hubertmis, 2017-03-13:
When I experimented with a SED device I noticed a small problem:
1. SED attaches to a parent.
2. Parent sends Data Response to the SED (new Pending Dataset).
3. SED sends Data Request to receive new Pending Dataset.
4. Parent puts Data Response with the new Dataset to the indirect queue.
5. SED is going to retrieve Data Response when poll timer expires. It can be in a few hours or days.

Maybe we could provide a procedure to notify MAC layer that MLE (or any other protocol) expects response. During waiting for a response the poll period could be much shorter than one set by an application.

This could be somehow a solution to #1353

GiedriusM, 2017-03-23:
How about adding a "expectedResponseTimeout" (exRT) field to otMessageInfo? By default, 0 would mean that no response is expected, and non-zero values - to expect a response in no more than exRT. MAC layer would then send N data requests every exRT/N milliseconds. Or instead of linear delay between data requests there could be an exponential delay like in CSMA.
Also, if multiple messages with exRT!=0 were sent (e.g. from MLE and app layer), then the MAC could use some intermediate values, so that the device would not spam too many data requests.

abtink, 2017-03-23:
Good point.
I am working on adding a new class `DataPollManager` which contains all the logic for handling of data polls and selection of polling interval (based on a combination of state of OpenThread and wishes of the app/user). I should have the PR for this soon. This can be used to address this issue.

jwhui, 2017-04-03:
Proposed fix in #1551.

PR #1551 (2017-04-04): Use fast poll after sending an MLE Data Request.
Resolves #1459.
closed 2017-04-04 by jwhui