openthread/openthread #1462: Locating types in public includes?

jwhui, 2017-03-14:
At the start of the OpenThread project, there were only two public include files: `openthread-types.h` and `openthread.h`.  As features were added to the OpenThread API, the set of public include files grew beyond those two files.  Most recently, #1173 further broke out APIs into feature-specific include files, based on feedback in #1100.

The current state of OpenThread is that we continue to have an [openthread/types.h](https://github.com/openthread/openthread/blob/master/include/openthread/types.h) that includes a large number of types.  At the same time, we have other public includes that also include types (e.g. [openthread/instance.h](https://github.com/openthread/openthread/blob/master/include/openthread/instance.h), and [openthread/udp.h](https://github.com/openthread/openthread/blob/master/include/openthread/udp.h)).

It seems we should be consistent about where to place types.  I see a few options moving forward:

1. Colocate all OpenThread types into a single [openthread/types.h](https://github.com/openthread/openthread/blob/master/include/openthread/types.h).
2. Break out types into their feature-specific headers.
3. A hybrid, where types for core features (those always compiled in) are included in something like a `openthread/core-types.h` and break out optional features (those with configure options) into feature-specific headers.

This issue is to collect thoughts and feedback from the OpenThread community.

nibanks, 2017-03-14:
My preference is to keep types closes associated with their features. That way, if a feature isn't included (not even the header is included) then no additional types get included either. No need to add additional complexity/confusion with unnecessary types.

pvanhorn, 2017-03-14:
One use case is coding for an NCP host, using code that can be recompiled to use Openthread on an SOC system.  What I mean by that is in one instance a firmware application would be resident on the same processor as Openthread and would interact with it through a middle layer module which hides whether Openthread is on the same processor (SOC) or a different processor (NCP).  The most efficient implementation for this application is to use Openthread types for all of the interactions between it and Openthread whether or not it is NCP or SOC.  The challenge is in identifying what subset of files must reside on the host in the NCP model in order for the system to build.  Currently, that set is limited to the following;
- types.h
- toolchain.h <required by types.h>
- code_utils.hpp <required by hdlc.cpp>
- spinel.c <provides api's for packing/unpacking UART frames>
- spinel.h
- hdlc.cpp <provides api's for  encoding/decoding UART frames>
- hdlc.hpp

The fewer files the better, as every time the system adopts a new firmware image for the NCP it must also adopt a new set of source files for what represents the host side footprint. Currently there are two files; toolchain.h and code_utils.hpp, that are brought in to satisfy other required files. If more header files are required, it is only reasonable to assume the list of indirect requirements would also increase.

For ease of supporting NCP hosts, I would ask that we make the set of required source files that a host must adopt remain as small as possible.  This of course means that I would like all of the openthread types to reside in a single types file with as few as possible #include directives inside.

It would also be useful if the set of files needed to build a NCP host in this fashion were pre-packaged into a directory either as part of make or perhaps some other operation.

pvanhorn, 2017-03-20:
For reasons stated above it would be helpful to me for all types to reside in a single header file or for all types to reside in a set of header files which only contain types and have as few #includes as possible.  Can we move forward with this approach?

PR #2946 (2018-08-08): [types] move types into specific headers
Resolves #1462.
closed 2018-08-08 by jwhui