openthread/openthread #1466: Switching from astyle to uncrustify.

jwhui, 2017-03-14:
The OpenThread project currently uses [astyle](http://astyle.sourceforge.net/) to automatically format code and provide baseline enforcement of the OpenThread code style.  In general, I've found astyle to be fairly limited in its flexibility.

This is a proposal to switch to [uncrustify](https://github.com/uncrustify/uncrustify).  For better (or worse), uncrustify supports many more options than astyle.  In short, astyle allows automated formatting and enforcement of more code style aspects, including:

- Left-aligned variable declarations.
- Aligned assignment operators.
- Left-aligned end-of-line comments.
- ... and many more.

aeliot, 2017-03-14:
Could we also get a code style guide?  Possibly a page on the wiki laying out the style guidelines of OpenThread?

jwhui, 2017-03-22:
Initial style guide was added with #1490.

closed 2018-02-02 by jwhui