openthread/openthread #1467: Changes to support building in IDEs

DuaneEllis-TI, 2017-03-14:
**Highlevel: #1** - Current autoconf, ./configure, Makefile.am, etc (collectively, auto-build-system) method of building things is not very compatible with GUI based IDEs

**Highlevel #2** - Support for non-GNU toolchains

**Highlevel #3** - Data structures across applications and libraries must be identical at link time.

**Problem:** In src/core/Makefile.am - lists numerous IF/ELSE/ENDIF which choose which files are added or not added to a specific library.  While IDEs can provide some level of exclusion, they are not that flexible. 

_Note:_ There are many other examples in other Makefiles through out openthread.

**Solution:** Always include all source files in all builds, but in the source code use #IF/#ELSE/#ENDIF constructs to disable entire files. In some cases this will result in the compiler producing an "empty-translation unit"

_Note:_  Some IDEs support exclusion based on 'configuration name', but each IDE does this a different way

**Problem:**  Need both --enable-cli-ftd --enable-cli-mtd, and likewise  --enable-mtd 

Need granularity such that  CLI can be built with MTD only,, And NCP can be built using MTD or FTD
Current ./configure solution is:  FTD for both, or FTD for Neither

**Solution:**  Add more ./configure switches.

Add --enable-cli-{yes|mtd|ftd|no} where YES = both
Add: --enable-ncp-mode={mtd|ftd|both}
Note that: --enable-ncp={spi|uart} is already taken.

**Problem:** Libraries that include 'downstream' header files should be built using the same flavor of downstream header files the libraries they are are linked against where built with.

_Example:_ The NCP-MTD-APP - linked with the 'agnostic-NCP- library, linked with the Openthread-MTD library fails both (1) whole program optimization and (2) static code analysis checks because:  The NCP-Library was built against only one of the two (MTD or FTD) header file configurations.

**Solution:** Require all libraries and all applications to affirmatively choose MTD vrs FTD at build time such that libraries, and thus internal headers are consistent across the entire build.

**Problem:** Some toolchains do not provide all functions or have incompatible implementations.

Some of this was addressed previously (header files only), but not completely (source implementations of missing functions)

_Examples include:_  "strnlen()" - Spinel code has s a private copy of strnlen()

**Solution:** 

Must be: "IDE Friendly"
Add:  include/openthread/platform/missing.h - for missing prototypes.
Add:  Missing functions to platform specific code.
FUTURE: Perhaps these could be moved to a more common location.

jwhui, 2017-03-14:
Thanks for this proposal!  While the OpenThread project uses the GNU toolchain to support open and free development, we also have a goal of being toolchain agnostic.  At a high-level, I'm supportive of these changes since they help us achieve that goal.

On supporting missing library functions, #737 is a long-standing PR to resolve this.  @darconeous, do you have plans to move that forward?  Would also appreciate your feedback on @DuaneEllis-TI's proposal above.

aeliot, 2017-03-14:
The `./configure` and autoconf way of building is fairly standard as far as I know, what IDE are you using that is having issues with it?  Is there an alternative build system that would provide the same functionality?

DuaneEllis-TI, 2017-03-14:
The definition of Standard is relative to your specific vantage point. ./configure is generally highly focused on the Gnu/Linux world which is command line dominate.

There are _many_ very good embedded engineers who exclusively use (or require, or their employer requires) an IDE, a subset of this population is very command line adverse.

TI for example - across our multiple chip families support these IDEs:

- TI-Code Composer Studio Using GNU tools 
- TI-Code Composer Studio Using the TI Compiler
- IAR Tools Using the IAR compiler
- Keil Tools Using ARM tools from ARM ( believe is, or this will become Clang )

If you speak with other chip vendors, they would have other combinations, and add their own tool solution to the mix.

-Duane Ellis

beriberikix, 2017-03-16:
Reiterating @jwhui, one of the goals of OpenThread is to be portability and allow silicon vendors to develop the best OpenThread experience for their silicon and toolchains. It's nice to have a LCD/baseline around open source tools (that everyone can run) but enable the flexibility for commercially-supported options. Big fan of this proposal!

jwhui, 2017-05-25:
Resolved by #1619.

closed 2017-05-25 by jwhui