openthread/openthread #1476: Compiler vrs Compiler issues, vrs IDE issues.

DuaneEllis-TI, 2017-03-16:
Open thread relies upon a number of standard C header files that are different across compilers, and compiler releases.

For example:

#include <string.h> - depending on the compiler does not have some functions.
This also varies by compiler version, (ie: a fix was introduce in a later version)

In the GNU Autoconf world this is solved via either  "libiberty' or various "#ifdefs"
GCC supports an   #include_next <filename> scheme - other compilers do not.

a) This does not play nicely with IDEs that do not support Autoconf.
b) This also makes code "messy" with lots of #if/#else/#endif fixes.

I propose the following solution:

If a standard header files needs to be fixed for a specific tool.
Then - the following action is taken.

Step 1:

A "pseudo-replacement" header file is created in a new directory called: "fixed"
    #include <openthread/platform/fixed/fixed_FILENAME.H>

Alternatively it could be placed in   ${openthread}/src/<somewhere>/fixed/fixed_FILENAME.H
The concern here is if thread ever provids a "C++" api - with functions defined within a class.

Note: specifically the original filename should not be used "as is", but a standard prefix or suffix be added to the original filename to make it unique. In this example I propose: "fixed_FILENAME.h"

Example:  #include <stdio.h>
Becomes:     #include <openthread/platform/fixed/stdio.h 

Step 2:

This new <openthread/platform/fixed/fixed_FILENAME.H>
Must eventually include the tool chain provided FILENAME.H

But also contains any required compiler/library fixes in the form of:  #if/#else/#endif

Step 3:  Problematic Thread code then includes the "fixed" header file in lue of the standard header file.

aeliot, 2017-03-17:
These standard headers, and the function signatures, should be the same across compilers that support C99 (ISO/IEC 9899), shouldn't they?  Since we are using C99, I think as long as we are carful to only use standard functions that are from C99 and not a later revision, we should be safe.

DuaneEllis-TI, 2017-03-17:
Question: Is strnlen() c99? or was this added at a later date 

I believe this is an online copy of the C99 standard:

http://cs.nyu.edu/courses/summer12/CSCI-GA.2110-001/downloads/C99.pdf

the function is not described in that document.

Other places I have tried/looked.

I do not see it here: http://en.cppreference.com/w/c/string/byte
I do not see it here: https://en.wikibooks.org/wiki/C_Programming/Strings
I do not see it here: https://www-s.acm.illinois.edu/webmonkeys/book/c_guide/2.14.html

aeliot, 2017-03-17:
It looks like ```size_t strnlen(const char *s, size_t maxlen);``` is actually a posix function.  If we are using it in the code then it should be replaced with a C99 compliant alternative.

aeliot, 2017-03-17:
So, for `strnlen` it looks like this issue has already been addressed, with spinel providing its own implementation of `strnlen` for compilers that dont provide one.

DuaneEllis-TI, 2017-03-17:
I don't think there exists a "C99" compliant version.

I believe that the idea of "strnlen()" is generaly from a security point of view. If the string is really super long (ie: buffer overflow attack) then stop now - do not search until the end of memory or crash.

the spinel solution does not solve the problem.

I don't think: https://github.com/openthread/openthread/issues/279 
completed/fixed the problem here - it addressed "spinel" only.

a) As Adam points out - an alternately named function is effectively implemented in  the NCP library  here: https://github.com/openthread/openthread/blob/master/src/ncp/spinel.c

b) Code such as:  

https://github.com/openthread/openthread/blob/master/src/core/mac/mac_frame.hpp#L913

Problem 1:  Does not have a header that it can include that provides this decl.

Problem 2: And unless an application links the NCP library, and uses that alternatively named function this will not solve the problem.

darconeous, 2017-03-20:
#737 is a somewhat unloved pull request that I need to get working again. It sets up a framework for pulling in implementations of functions which aren't present on some platforms. It could be easily extended to include `strnlen()`.

Unfortunately that PR hasn't gotten a lot of love since I first proposed it, and I need to do some additional work on it. It also doesn't really take into considerations IDEs.

Keep in mind that these "replacement functions" aren't really intended to be exposed to the applications, they should only be exposed for OpenThread internal use only.

The Spinel `strnlen()` was just a quick implementation of a function that was conveniently small to implement. It is statically defined and so obviously not intended to be exported out of the file it is used in.

DuaneEllis-TI, 2017-04-20:
@jwhui, @darconeous, and @nibanks 

I'd like to draw your attention to this issue, it is the next one I would like to better address.

This is the next large item I want to push into OpenThread.

Code in openthread currently does this:

`      #include <string.h>
`
To fix missing/problems with string.h - and other related, platforms do some "Gymnastics"
In ./configure land, we "force feed" header files into the compiler.
In windows-land, the openthread-windows-config.h file includes these files.

Doing this header file force feed into the compiler gets ugly in IDE land. Its doable, but ugly.

Another aspect of this is the skill-set of the target developer, some of us propeller heads know well the internal workings of the compiler, and the ide configuration files, etc...  In the broad market, not all developers really understand this, especially noobs, if the goal is to have a mass market for thread - then doing things that are easy and quick to understand, is the better way.

I don't think the gymnastics approach used right now is a good long term solution, and thus I want to put forth what I think is a better choice or method

It comes down to this:

**STEP 1**

OLD CODE: 
```
#include <string.h> /* old code */
```
NEW CODE
```
#include <utils/wrap_string.h> /* use this name instead */
```
The same would apply to  stdbool.h, and stdint.h for windows, and more importantly - the same model can be used for **ALL** future files in the same way, just create "wrap_SOMENAME.h" as needed.

In a perfect world, this new "wrap_string.h" - just includes <string.h>
But we don't live in that world.

An example implementation of 'wrap_string.h' might be like this:

Note: I have this ready to go now, we are using this internally now.
My only questions are, if I push this up, how should this be handled

a)  Separate library? if so what name?
    This would obvious need a bit of tweaking on windows side (a new library to make)

b)   Or integrating into the existing openthread library
    I think this is the correct/better choice, it is more transparent.
   That said, it would require add/remove a few files to the vcxproject files.

c) Or nope, this is not the correct way to solve this.

```
/* copyright statement here */
#if !defined(WRAP_STRING_H)
#define WRAP_STRING_H

#include "wrap_common.h" // common wrapper macros */

/* include the system provided header */
#include <string.h>

/* these are the alternate implementations */
/* See: https://www.freebsd.org/cgi/man.cgi?query=strlcpy */
WRAP_EXTERN_C size_t missing_strlcpy(char   *dst, const char *src, size_t dstsize);
/* See: https://www.freebsd.org/cgi/man.cgi?query=strlcat */
WRAP_EXTERN_C size_t missing_strlcat(char   *dst, const char *src, size_t dstsize);
/* See: https://www.freebsd.org/cgi/man.cgi?query=strnlen */
WRAP_EXTERN_C size_t missing_strnlen(const char *s, size_t maxlen);

/* undefine any compiler supplied function. */
#undef strlcat
#undef strlcpy
#undef strnlen

/* Define these, it just works always by default ... */
#define strlcat missing_strlcat
#define strlcpy missing_strlcpy
#define strnlen missing_strnlen

/* Undefine below - as [er platform */

#if defined(_MSV_VER)
#undef strnlen /* provided by visual studio */
#endif

#if _wrap_ti_arm
/* TI_ARM compiler is missing all */
#endif

#if __linux__
#undef strnlen /* provided by gcc */
/* strlcpy - missing */
/* strlcat  - missing */
#endif

#if _wrap_iar_arm
/* IAR systems for ARM  */
/* does not supply any in all cases */
#endif

#endif  // WRAP_STRING_H

```

darconeous, 2017-04-20:
I'm not opposed to it. Seems like a workable and convenient solution.

darconeous, 2017-04-20:
Regarding the library... I would assume this stuff would just go with the platform bits. I'm guessing that is option B? I would think we certainly wouldn't want to expose these headers to applications linking against us.

DuaneEllis-TI, 2017-04-20:
Technically it is doable (putting the library in platform) but currently things in "src" do not include  headers from the platform directory, and there is no "platform include" directory. Is that what we want? I don't think so. 

If we want these private to openthread, I think the headers should be inside the ${openthread}/src directory somewhere.

The question is really this:  Do we want to require linking another library to link against? Yes or No.

Sort of:  Always link:  ${LIB_OPENTHREAD_PART1}  and ${LIB_OPENTHREAD_PART2}

That part2 library would have the missing string functions...  I think this is awkward and weird, it is btw - what I have currently implemented internally. I'm not married to "two-libraries" I am questioning if two libraries is the correct choice.

Hence I think Option (B) - but part of: "libopenthread" [not platform] is the place it belongs.

jwhui, 2017-05-13:
Resolved by #1642.

closed 2017-05-13 by jwhui