openthread/openthread #1477: Win App Logo Change

aeliot, 2017-03-17:
The windows app currently uses the Thread Group logo. (https://github.com/openthread/openthread/tree/master/etc/visual-studio/Assets)
These logos should probably be changed to the OT logo, since this app is by OpenThread and not by the Thread Group.

nibanks, 2017-03-17:
Can someone point me to an image file of 'the OT logo'? As indicated in the link above, we need multiple sized, from a small icon to a splash screen.

aeliot, 2017-03-17:
The full OpenThread logo can be found in the docs folder, I'm working on getting the sort OT logo

beriberikix, 2017-03-17:
[image]

beriberikix, 2017-03-17:
Here's a high-res logo. I have it as a .ai if you want multiple sizes, just need to know the resolutions.

nibanks, 2017-03-17:
I need the following sizes:

Icon
- 24 x 24
- 48 x 48
- 50 x 50
- 88 x 88
- 300 x 300

Splash Screen
- 620 x 300
- 1240 x 600

beriberikix, 2017-03-17:
Do you want the "OT" logo for all sizes?

nibanks, 2017-03-17:
I'm fine with either that or the 'OpenThread' logo for the Splash Screen images.

beriberikix, 2017-03-18:
I haven't spent this much time in Illustrator in a while...does these work?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

beriberikix, 2017-03-20:
[winapp_images.zip](https://github.com/openthread/openthread/files/855579/winapp_images.zip)

Adding as a .zip as well.

nibanks, 2017-03-20:
The logos look pretty good except with the default color when pinned to the start menu. Looks good when pinned to task bar. 

[image]

[image]

turon, 2017-03-20:
@nibanks Is there a way to specify that white should be used as the default background on Win desktop?  The .png is just defining the transparent parts, so the background color is likely metadata of some sort in the desktop config.

nibanks, 2017-03-20:
I updated my PR #1489 use a white background.

aeliot, 2017-03-21:
@nibanks could you also update the screenshots in /doc/images?  The following still have the old background:
windows-app-interface-list.png
windows-app-talk-client.png
windows-app-talk-server.png

jwhui, 2017-03-29:
Addressed by #1529.

closed 2017-03-29 by jwhui