openthread/openthread #1495: Radio is in receiving mode while Thread is not started

GiedriusM, 2017-03-21:
When mRxOnWhenIdle is false and (if) Thread interface is not started, the radio does not enter sleep mode. In the example code given below, otPlatRadioReceive is called initially, however otPlatRadioSleep is never called. I consider Thread to be "idle" while it is in kDeviceRoleDisabled state and therefore it should put the radio into sleep mode. IMO, in this scenario otPlatRadioReceive should not be called at all, instead, radio should be initialized into the low-power sleep mode.
As an example, for me this becomes relevant when a battery powered device fails to find a commissioned network and wants to enter low-power sleep mode (i.e. to retry after some time), however the radio is kept in RX state and wastes battery energy.

```C
otLinkModeConfig mode = {0};
mode.mRxOnWhenIdle = 0;
mode.mSecureDataRequests = 1;
mode.mDeviceType = 0;
mode.mNetworkData = 0;

otSetRouterRoleEnabled(sInstance, false);
otSetLinkMode(sInstance, mode);
otInterfaceUp(sInstance);

while (1) {
    otProcessQueuedTasklets(sInstance);
    platformProcessDrivers(sInstance);
}
```

jwhui, 2017-03-21:
The Device Mode is a Thread-level concept (I agree that `otLinkModeConfig` is named inappropriately and should be called something like `otThreadDeviceMode`).  That said, the power management of the radio is only managed by the Thread layer when Thread is enabled.

We have use cases where the radio needs to be enabled to communicate IPv6 link-local messages while Thread is disabled.  The separation between [`otIp6SetEnabled`](https://github.com/openthread/openthread/blob/master/include/openthread/ip6.h#L67) and [`otThreadSetEnabled`](https://github.com/openthread/openthread/blob/master/include/openthread/thread.h#L78) allows us to support those use cases.  When we desire to disable the radio completely, we disable both IPv6 and Thread.  Is this a workable solution for you?

Note that the APIs were recently changed in #1462.

GiedriusM, 2017-03-22:
Yes, this and few other ways I've found works as solution/workaround. However I should've been more clear and include a question whether this an intended behavior or a bug. Previously I though of mRxOnWhenIdle=false (a.k.a. "sleepy-mode") as 802.15.4 feature, but from your answer I make a conclusion, that this is Thread-only feature and that the Thread network expects the joiner device to be "always on" and there is no ability for that device to enter sleep mode (while joining). Am I correct?

jwhui, 2017-03-22:
Yes, you are correct.  The rx-on-when-idle device mode is managed at the Thread level.  When a device is joining as part of Thread Commissioning or performing an MLE attach, the radio's receiver is always enabled (except when transmitting).  The rx-on-when-idle mode only kicks in when a device is attached to a parent.

GiedriusM, 2017-03-22:
Thank you for the explanation, it makes much more sense now.

closed 2017-03-22 by GiedriusM