openthread/openthread #1531: nrf52840 RTT Logging Not Working

nibanks, 2017-03-29:
Has something changed with the nrf52840 logging recently? I am following the steps [here](https://github.com/openthread/openthread/tree/master/examples/platforms/nrf52840#logging-module), but no logs ever show up.

I changed the default log level to `OPENTHREAD_LOG_LEVEL_DEBG` and enabled `OPENTHREAD_CONFIG_LOG_PLATFORM`, and just built the default image (both NCP & CLI). I have installed both NCP & CLI on different devices, and individually connected the J-Link RTT Viewer. Nothing ever shows up in the 'All Terminals' windows.

[image]

[image]

[image]

Any suggestions here?

LuDuda, 2017-03-30:
I have just performed quick smoke test on the current master:

```
$ sed -ie 's/OPENTHREAD_LOG_LEVEL_CRIT/OPENTHREAD_LOG_LEVEL_DEBG/g' src/core/openthread-core-default-config.h
$ ./bootstrap && make -f examples/Makefile-nrf52840
$ arm-none-eabi-objcopy -O ihex output/nrf52840/bin/ot-cli-ftd cli.hex
$ nrfjprog -f nrf52 --chiperase --program cli.hex
$ nrfjprog -f nrf52 -r
```

I installed the same version of Segger J-Link (6.12a) and updated the firmware on the board to use also 6.12a firmware (by J-Link Configurator). I also used same settings as you have (including speed), and it appears to work fine on Windows.

[image]

[image]

Some ideas what could went wrong:
1) Your J-Link firmware on nRF52840 PDK is much older than your version on Windows - use J-Link Configurator to align both versions.

2) I see that there is a tab called "Terminal 0" in your RTT Viewer. That means that J-Link found RTT buffer on nRF board, because of that i would guess that logging level may be not correctly set or DEFAULT_LOGGING is used?

3) Turn RTT Viewer off while programing a board.

4) It should not be related but there is a problem in Segger firmware that prevent to send data over virtual COM - but what i know only on virtual machines. The solution is to disable mass storage:

    - Run JLink Commander
    - Choose default setting nRF52832_xxaa and SWD
    - Exectute **MSDDisable**

Give me know if any of above works? We may update README.md to address it.

nibanks, 2017-03-30:
I just restart my Windows machine, rebuilt and reflashed the devices and now everything seems to work. Thanks for double checking!

closed 2017-03-30 by nibanks

turon, 2017-03-30:
Hi Nick,

I also had some issues with the logging recently.

The solution was to use a version of softdevice matched to what is
supported by the drivers in the openthread tree.

I'll let the Nordic folks provide more definitive details.

Martin

_____________________________
Martin Turon  |  Nest Labs