openthread/openthread #1533: SPINEL - protocol concerns

DuaneEllis-TI, 2017-03-30:
some issues with the SPINEL protocol need to be addressed, specifically these deal with power management and sleepy devices.

## Issue 1 - the NCP may be in a deep sleep.

The NCP device may be asleep, or in an ultra low power state and not ready to respond quickly  (immediately) to a host that may be transmitting data.

Typically a GPIO signal from the HOST to wake, or draw the attention of the slave.

The simple approach would be to think the SPI-CS signal is the solution. The reason this fails is many SPI hosts will drop the CS to active, and immediately (within ZERO micro seconds) begin transmitting (clocking) data out of the SPI port. 

Depending upon the sleep state of the NCP and the wakup timing requirements the NCP may not be able to respond, and configure it self for the SPI transfer prior to the host transmitting (or perhaps the UART baudclock source needs to be turned back on)

Examples include:  Interrupt servicing over head at "slow (ie: 32khz) sleep clock rate", length of time to re-enable the on chip oscillator, startup the external crystal and walk the internal power regulators back up to operating voltages.

This can often be made worse when the SPI interface is operating at 10mhz, or higher (vrs the 32khz sleep clock the NCP is waking up from).

The result is the Host & NCP slave can be "off-by-a-bit" or two, or three during the transfer and this problem only occurs rarely... and nobody can figure out why or when this occurs.. maybe the chip wakes up slower when it is very cold (think sensor inside a freezer, or a pool sensor in midwinter)

What is missing is a "NCP-IS-READY" hand shake back to the host.
(total 2 pins)

## Issue 2 How fast can the HOST wake up?

The NCP may need to wake the host up and tell the host there is data to be sent.

This slightly discussed in the SPINEL protocol and repeated here for completeness.

What is not considered is this: A battery powered Linux host might be asleep and in a low power state (example: A cell-phone-data-only device connected to an NCP) the NCP cannot just "raise the request" and start transmitting out the UART... that host might also need to "wake up" and turn clocks on, enable PLLs and so forth before the UART baudrate clock is operational.

Typically another GPIO pair (ncp to HOST) is used to indicate this request.
total: 4 pins.

## Issue 3 - Timing (latency) Problems 

COST PROBLEM - every pin on a chip costs money, or makes a product require a more expensive chip with more pins to provide the solution (this is great for those selling the chip, but bad for buyer of the chip)

A fundamental truth is this:  Digital signals can cross 3 to 4 inches on a circuit board without problems at relatively high frequency. There is a myth that UARTs are not reliable, that was true when the signal ran across 300 feet of wire from one end of a factory to another end of the factory. Here we are dealing with 3 to 4 inches of signal traces on a circuit board or small flex cable. The state of the art is such that you do not have signal quality problems. The truth is often something else. The problem is often the power up sequences (above) or this race condition, or you are writing very in-efficient C code.

In general, one does not want 4 GPIO hand shake pins 

- To solve issue 1  (host -> ncp) requires 2 pins, a request and ack
- To solve issue 2 (ncp -> host) requires 2 pins, a request and ack.
- 

The economic need is this:
    2 pins UART (txd/rxd)  + 2 GPIO handshake = total 4
    3 pins SPI  (mosi,miso,clock)  + 2 GPIO handshake = total 5

Notice that the SPI does not use an actual CS signal - in theory it is present, just not hooked up or the GPIO pin is reconfigured in a "non-SPI-mode" and is used as a GPIO pin instead. 

If the slave (NCP) requires this signal, the HOST_REQACK can be used as the SPI_CS signal. The slave will see the SPI_CS assert, then a seemingly long delay before the SPI_CLK starts.

## Issue 4  - hardware design problem, power management

Sometimes the GPIO signals are also used to enable power to voltage level shifters between different power domains (ie: maybe the HOST uses VDD-IO as 1.2v, but the radio uses 3.3v?) Will enabling the level shift cost another GPIO pin? Or can we reuse one of the HANDSHAKE GPIO signals? The Money answer is simple: An existing GPIO signal needs to be reused reused.

## The Timing Race Problem that occurs because of issues 1, 2, 3 and 4.

The race condition happens because of economizing the number of GPIO signals

The race condition is this:

## At Time 0 - 
Both the Host and the NCP decide "i-need-to-send-data"
Both assert their single GPIO signal at the exact same time.
  
the host is often a linux machine, the user level IRQ latency is horrible compared to the NCP-microcontroller. Latency is defined as:  NCP-GPIO assert, to HOST-GPIO-Ready assert, or further first bit has transferred over the connection.

## The RACE is this:

[... 70 lines cut ...]

The host side, from user space through the kernel and out the GPIO driver is slow. We can use this to our advantage.  

First: Measure latency. 

Write a simple user space 'pulse-the-gpio' application on Linux and measure fastest time an 
The code needs to perform a WRITE pin (X [toggling the pin]), then READ pin (Y). Measure the period of the wave form, this is 2x the fastest speed the board change a GPIO Pin.  Note: Linux (and windows) user space GPIO is rather slow. We will use this to our advantage.

Second: 

Pick 50% of that period - the NCP must meet this timing requirement: From END OF DATA to SLAVE_REQACK de-assert - must be completed in that (50%) time window.

OR - The host must insert a busy wait( as described below )

The host at the end of any transfer:

- step 1 - write the HOST_REQACK pin, 
- step 2 - if required insert a short sleep (often this is not required)
- step 3 - read the SLAVE_REQACK pin
- step 4 - determine: Loop for more data, or wait for another interrupt.
- Note: this technique is often found in UART drivers where the hardware supports a FIFO buffer, you will notice the actual hardware IRQ contains a loop - that drains the fifo. That is effectively what we are doing here, the loop draining the transfer buffers.
- 

Note that a Microcontroller based HOST is often much faster at raw GPIO io then a host.
It does vary greatly due to the architecture of the host chip ( often a 2ghz android device cannot read the GPIO pins at 2ghz... they instead require/use a bus frequency shift to a lower speed IO bus)

darconeous, 2017-03-30:
It isn't entirely clear what you are proposing overall, so I'll go over individual statements below.

By the way, Spinel isn't acronym, so you can just call it "Spinel", not "SPINEL".

> The simple approach would be to think the SPI-CS signal is the solution. The reason this fails is many SPI hosts will drop the CS to active, and immediately (within ZERO micro seconds) begin transmitting (clocking) data out of the SPI port.

This isn't actually a problem with [the recommended Spinel SPI protocol](https://rawgit.com/openthread/openthread/master/doc/draft-spinel-protocol.html#rfc.appendix.A.2). If the slave is just waking up and isn't ready to transmit, the slave would keep MISO high. This signals to the master that the slave is throttling. The master will then either poll (very dozen milliseconds or so) and/or wait for the interrupt pin to be asserted.

Granted, this could likely be better documented.

> The result is the Host & NCP slave can be "off-by-a-bit" or two, or three during the transfer and this problem only occurs rarely... and nobody can figure out why or when this occurs.. maybe the [...]

I emphatically disagree with any recommendation that requires more pins: It is simply not necessary.

If you and the host are "off by a bit or two", then the problem is in your implementation. If the NCP wakes up from CS being asserted and it can't guarantee that it can handle the transaction then it should keep MISO high and ignore the frame and wait for the next one. The master will then prepare another frame that will start at the next slave-to-master interrupt or wait a dozen (or so) milliseconds and try again (whichever is sooner). No harm, no foul.

> The NCP may need to wake the host up and tell the host there is data to be sent.
>
> This slightly discussed in the SPINEL protocol and repeated here for completeness.
>
> What is not considered is this: [...]

Traditionally, we have done this using flow control and BREAK signals. When the host goes to sleep, it asserts flow control. When the NCP has something to send to the master while flow control is asserted, it will occasionally send a UART "break" until flow control is no longer asserted, at which point frames are transmitted as normal.

The same mechanism can be used with software flow control (requiring only two pins), as long as the underlying hardware treats BREAK as a specific UART signal, but this may be rather tricky to get right.

Another way that this could be handled is with a serial reliability layer. This is nice because flow control is implicit and wouldn't require additional pins (at the cost of some additional wakeup latency). But this isn't yet specified.

I agree this needs to be better elaborated on in the appendix. We also haven't implemented this yet in the OpenThread NCP, so we need to go ahead and do that.

> The economic need is this:
>2 pins UART (txd/rxd) + 2 GPIO handshake = total 4
>3 pins SPI (mosi,miso,clock) + 2 GPIO handshake = total 5

We use TXD, RXD, CTS, RTS for UART, using the signaling scheme I described above. Works great. We could define a mechanism which only uses TXD and RXD, but for that we would need a reliability and retransmission layer which hasn't been specified.

For SPI we use MOSI, MISO, CLK, CS, and INT. We also use RESET. Theoretically RESET isn't required, but we have found that it is useful for the master to have the ability to force a hardware reset of the slave out-of-band. Also, theoretically even INT isn't required, since you can always poll with the current SPI protocol, further reducing pin count.

> Notice that the SPI does not use an actual CS signal

It seems like you are proposing that we change the SPI protocol to require a two-way handshake before the master starts clocking out bits, something like Master-Ready and Slave-Ready, where if at any point one side wants to start a transaction, that side asserts its pin and then waits for the other side to assert their pin. The signal of asserting CS is then effectively the logical AND of the state of both pins (at least for the starting edge), where de-asserting CS is (presumably) just the falling edge of Master-Ready.

I don't like this system for the following reasons:

* It doesn't solve any problem that isn't already sufficiently solved by the current recommended protocol
* It makes the implementation that drives the master from Linux user-space more complex.
* It prevents (or reduces the efficiency of) sharing the SPI bus with other peripherals.

We are using the Spinel-recommended SPI protocol right now and it works great.

That being said, I think the documentation could use improvement, which we are working in internally.

DuaneEllis-TI, 2017-03-30:
>> This isn't actually a problem with the recommended Spinel SPI protocol. If the slave is just waking up and isn't ready to transmit, the slave would keep MISO high. 

Effectively the MISO pin becomes another handshake signal.

An important thing to to note is handshake signals may be active low, or active high depending upon the actual HW architecture in the system.  Active high signals (ie: 0V is the idle state) are more power hungry only due to parasitic leakage in the HW design, and very anti-battery life. A few micro amps integrated over a long sleep time adds up. To paraphrase Yogi Berra - A few microamps here, and a few microamps there now you are talking about real money. Another problem you can have is the VDD-IO voltage rails may change level in sleep mode, remember power (watts) is a driven by voltage (e) more specifically e^2, so being able to drop voltage during sleep is important,  and 0^2 is always 0.  Changing VDD-IO presents other problems if the signal is active low, where as an active high signal does not. Momentary over voltage in those situations is cheaply solved with an inline resistor.

Some SOCs provide pin-observation at all times, others do not, which means: Some hosts must dynamically reconfigure the MISO pin, which is not always simple from user space (ie: Linux Device tree nonsense)

>> [in the uart case] When the NCP has something to send to the master while flow control is asserted, it will occasionally send a UART "break"

Interesting concept...  In my experience UART break detection does not work very well across all the micro controller that I have used in projects. Some do well, some do horribly it is very hit or miss.  Thus I have always avoided this.

>> Theoretically RESET isn't required, but we have found that it is useful for the master to have the ability to force a hardware reset of the slave out-of-band.

Agreed - also needed for UART mode to do 'in-place' flash update of the device from the linux host.

>> It seems like you are proposing that we change the SPI protocol to require a two-way handshake

yes exactly. This is also needed for UART mode. I probably was not very clear.

>> It makes the implementation that drives the master from Linux user-space more complex

not really, it works like this:

# Send Steps

Send Step 0:
  DO NOT assert READY yet
  Perform a MUTEX lock in the IO scheme.
  DISARM the RX request logic (ie: edge irq)

  Power up IO circuit, 
  Enable clocks in all required power domains.

SEND Step 1:
  sender  assert MY_READY  (master or slave)
  (set GPIO pin)

SEND Step 2:
  sender wait    OTHER_READY
  (poll GPIO pin)

SEND Step 3:
  Transfer is performed (uart or spi)
  In spi case the Linux host starts clocking the SPI_CLOCK pin.
  In linux user space this is just a call to the SPI transfer function.

SEND Step 4:
  to handle race condition:
      Sender must examine other reply
      in UART case - if no bytes - no message
      in SPI case - if all bytes = STUFFING value no message

  If "other_message exists" and is not yet complete
  The host must complete this message:

  Example SPI case
       HOST only clocked in/out 20 bytes, 
       And the other message is longer...
       The SPI interface will not complete the transfer
       The host must send "stuffing values" to produce transfer clocks.

  Example UART case
       typically nothing to do here
       What you don't want to do is shut down the IO power yet.

   If the message was completed - post the message into the RX thread.

SEND Step 5:
   REARM the RX request logic, this must occur before dropping request.
   Sender DROPS request.

SEND step 6:
    Note: Do not wait until receiver Drops READY
    This might never happen, you might be too slow and miss this.

SEND step 7:
    Release Mutex lock on the IO scheme

SEND step 8:
     Done.

# Rx Step

RX STEP 1:  Wait for REQUEST signal..
       Note this is an *EDGE* request, not a level request.

RX STEP 2:   MUTEX LOCK transfer scheme.

RX STEP 3:  Power up IO circuit, enable clocks, set IO voltages.

RX STEP 4:   Assert READY (this is the request to other side)

RX STEP 5:  Wait for others to assert ready.
       
RX STEP 5:  Transfer is performed.
        If using SPI, the the receiver must send "stuffing values" 
        Either all FF or all 00

RX STEP 6:  End of transfer
       RE-ARM rx-edge detect.
       DROP ready signal.
       Release Mutex

RX STEP 7:
       Post RX message received.

darconeous, 2017-03-30:
> Effectively the MISO pin becomes another handshake signal.

More like the contents of the MISO frame are used as a handshake signal.

> An important thing to to note is handshake signals may be active low, or active high depending upon the actual HW architecture in the system. Active high signals (ie: 0V is the idle state) are more [...]

The current implementation of the SPI protocol in `spi-hdlc-adapter` makes no differentiation between the slave sending a frame full of 0x00 bytes or 0xFF bytes. Both will induce the host simply try again later. At sleep, the slave would pull MISO to whatever state is the lowest power. At wakeup, the slave would boot up and then wait for CS to no longer be asserted before initializing the SPI component.

> Some SOCs provide pin-observation at all times, others do not, which means: Some hosts must dynamically reconfigure the MISO pin, which is not always simple from user space (ie: Linux Device tree nonsense)

Why in the world would the host need to reconfigure the MISO pin? The host just gets the frame. If it is all 0xFFs or all 0x00s, then it knows the slave is busy. No need to reconfigure the MISO pin.

This isn't hypothetical code that needs to be written. The Linux host side works Right Now™ with the scheme I just described.

> Interesting concept... In my experience UART break detection does not work very well across all the micro controller that I have used in projects. Some do well, some do horribly it is very hit or miss. Thus I have always avoided this.

Which is why I pointed out the XON alternative, for when hardware doesn't do BREAKs well.

I personally implemented the deep sleep and wakeup signaling on both our shipping UART-based products and shipping SPI-based products. I designed the recommended Spinel framing protocols with that experience in mind. Now I'm not perfect, I certainly have been known to make mistakes, so I'm more than willing to consider that there may be better alternatives. But so far I still do not understand what problem you are trying to solve that can't be easily solved using simple adjustments to what is already specified, implemented, and working.

DuaneEllis-TI, 2017-03-30:
duane>Why in the world would the host need to reconfigure the MISO pin? 
darconeous> Why in the world would the host need to reconfigure the MISO pin?

As I said, some SOCs do not provide pin observably.  

Your suggestion was the SPI-SLAVE holds MISO as an additional handshake. The problem occurs on SPI-MASTER side, example: PIN(X) is in mode GPIO, or PIN(X) is in mode (MISO).  For some SOCs when the pin is in a NON_GPIO mode reading the raw pin value (ie: as if it was a GPIO) is not possible due to the way the hardware was designed.

How then will the host know the slave is ready?

darconeous, 2017-03-30:
> As I said, some SOCs do not provide pin observably.

We observe the pin by looking at the MISO frame after the transaction.

darconeous, 2017-03-30:
If the slave holds MISO low, the frame received by the master will be all zero bits. If the slave holds MISO high, the frame received by master will be all one bits. There is absolutely no reason to change any pin settings on the master.

darconeous, 2017-03-30:
And it isn't really a handshake. It is a refusal by the slave to handle the frame.

DuaneEllis-TI, 2017-03-30:
How can we avoid this refusal?

darconeous, 2017-03-30:
We don't avoid it. Refusals are normal: they are how the slave asserts flow control.

The master tries again a dozen (or so) milliseconds later, or we try again immediately if interrupt gets asserted (which it likely would be if the NCP is just waking up).

There are two ways the slave can refuse a frame from the master:

* Passive refusal: Don't wiggle MISO. Leave it in the same state for the duration of CS being asserted.
* Active refusal: Clock out a well-formed 5-byte header on MISO, with a `RECV_LEN` of zero.

darconeous, 2017-04-19:
Closing. Please re-open if you have additional concerns.

closed 2017-04-19 by darconeous