openthread/openthread #1535: MeshForwarder::LogIp6Message not building

Gr3yh0und, 2017-03-30:
Current master is not building the LogIp6Message function when OPENTHREAD_CONFIG_LOG_LEVEL >= OPENTHREAD_LOG_LEVEL_INFO, which in my case is 4 > 3. Probably introduced in  #PR1515 ?

Compiler spits out:
>   CXX      thread/libopenthread_mtd_a-mesh_forwarder.o
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp: In member function 'void Thread::MeshForwarder::LogIp6Message(Thread::MeshForwarder::MessageAction, const Thread::Message&, const Thread::Mac::Address&, ThreadError)':
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2286:14: error: variable 'checksum' set but not used [-Werror=unused-but-set-variable]
>      uint16_t checksum = 0;
>               ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2289:10: error: unused variable 'stringBuffer' [-Werror=unused-variable]
>      char stringBuffer[Ip6::Address::kIp6AddressStringSize];
>           ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp: At global scope:
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2283:49: error: unused parameter 'aAction' [-Werror=unused-parameter]
>  void MeshForwarder::LogIp6Message(MessageAction aAction, const Message &aMessage, const Mac::Address &aMacAddress,
>                                                  ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2283:103: error: unused parameter 'aMacAddress' [-Werror=unused-parameter]
>  void MeshForwarder::LogIp6Message(MessageAction aAction, const Message &aMessage, const Mac::Address &aMacAddress,
>                                                                                                        ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2284:47: error: unused parameter 'aError' [-Werror=unused-parameter]
>                                    ThreadError aError)
>                                                ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp: In member function 'void Thread::MeshForwarder::LogIp6Message(Thread::MeshForwarder::MessageAction, const Thread::Message&, const Thread::Mac::Address&, ThreadError)':
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2286:14: error: variable 'checksum' set but not used [-Werror=unused-but-set-variable]
>      uint16_t checksum = 0;
>               ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2289:10: error: unused variable 'stringBuffer' [-Werror=unused-variable]
>      char stringBuffer[Ip6::Address::kIp6AddressStringSize];
>           ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp: At global scope:
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2283:49: error: unused parameter 'aAction' [-Werror=unused-parameter]
>  void MeshForwarder::LogIp6Message(MessageAction aAction, const Message &aMessage, const Mac::Address &aMacAddress,
>                                                  ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2283:103: error: unused parameter 'aMacAddress' [-Werror=unused-parameter]
>  void MeshForwarder::LogIp6Message(MessageAction aAction, const Message &aMessage, const Mac::Address &aMacAddress,
>                                                                                                        ^
> /home/gr3yh0und/Downloads/openthread/examples/../src/core/thread/mesh_forwarder.cpp:2284:47: error: unused parameter 'aError' [-Werror=unused-parameter]
>                                    ThreadError aError)
>                                                ^
> cc1plus: all warnings being treated as errors
> Makefile:3223: recipe for target 'thread/libopenthread_mtd_a-mesh_forwarder.o' failed
> make[4]: *** [thread/libopenthread_mtd_a-mesh_forwarder.o] Error 1

nibanks, 2017-03-30:
My PR #1532 just merged. It might have fixed your issues.

Gr3yh0und, 2017-03-30:
Just made sure I merged your commit into my branch, which I did earlier. When I build the current master for the CC2538DK and change the Debug level from `CRIT` to `DEBG` it breaks.

nibanks, 2017-03-30:
I just rebased my branch to the latest, and I have the following local changes in `openthread-core-default-config.h`:

[image]

[image]

I ran the following commands:

```
./bootstrap
make -f examples/Makefile-cc2538 clean
make -f examples/Makefile-cc2538
```

Everything seemed to build successfully.

Gr3yh0und, 2017-03-30:
Thanks Nick for checking that again. I did the exact same thing before... strange. It now does somehow build so maybe just another `clean` did the trick... I'm closing that until I can reproduce it...

Gr3yh0und, 2017-03-30:
Could you please try to disable MAC logging?

`#define OPENTHREAD_CONFIG_LOG_MAC                               0`

jwhui, 2017-03-30:
I'll have a PR out for this soon.  Thanks for catching this.

jwhui, 2017-03-30:
Submitted #1538.  Please review.

PR #1538 (2017-03-30): Only enable `LogIp6Message` when `OPENTHREAD_CONFIG_LOG_MAC` is 1.
Resolves #1535.
closed 2017-03-30 by jwhui