openthread/openthread #1536: Running NCP on Latest Wpantund failed.

rmadhuraj, 2017-03-30:
Hi ,
      I Recently updated the wpantund from 0.07.0 to 0.08.0. i.e., as shown in 
https://github.com/openthread/wpantund/blob/master/INSTALL.md#wpantund-installation-guide
And followed the simulator tutorial 
https://github.com/openthread/wpantund/wiki/OpenThread-Simulator-Tutorial
Now i'm unable to Ping with NCP or CLI node from LINUX terminal.

These are the steps i followed.

**For building:**

cd <openthrea-master>
$ ./bootstrap 
$ sudo ./configure --enable-ftd --enable-ncp --enable-cli --disable-docs --disable-tests --with-examples=posix 
$ sudo make

**on terminal 1 Running CLI instance**
$ cd /openthread-master/ 
$ examples/apps/cli/ot-cli-ftd 1
 panid 0x1234
Done
> ifconfig up
Done
> thread start
Done
> state
leader
> ipaddr
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:6000
fdde:ad00:beef:0:a7ac:d8b:2b3:456a
fe80:0:0:0:b4a5:cffc:eb42:f575
Done

**on term 2 running NCP instance**
$ cd ~/openthread-master/ 
$ sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp-ftd 2" -o SyslogMask " -info" -o Config:TUN:InterfaceName wpan0

wpantund[18422]: Starting wpantund 0.08.00 (Mar 30 2017 21:46:31) . . .
wpantund[18422]: 	BUILD_VERSION = f911961-dirty
wpantund[18422]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[18422]: Configuration file "/etc/wpantund.conf" read.
wpantund[18422]: CONFIGURATION PROPERTY "SyslogMask" IS DEPRECATED. Please use "Daemon:SyslogMask" instead.
wpantund[18422]: set-config-param: "Config:NCP:SocketPath" = "system:examples/apps/ncp/ot-ncp-ftd 2"
wpantund[18422]: set-config-param: "Config:TUN:InterfaceName" = "wpan0"
wpantund[18422]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[18422]: Ready. Using DBUS bus ":1.25"
wpantund[18424]: About to exec "examples/apps/ncp/ot-ncp-ftd 2"
wpantund[18422]: Running as root without dropping privileges!
wpantund[18422]: [-NCP-]: NCP was reset (STATUS_RESET_POWER_ON, 112)
wpantund[18422]: State change: "uninitialized" -> "offline"
wpantund[18422]: NCP is running "OPENTHREAD/0.01.00; none; Mar 30 2017 17:33:39"
wpantund[18422]: Driver is running "0.08.00 (/f911961-dirty; Mar 30 2017 21:46:31)"
wpantund[18422]: Unsuccessful fetching property "PROP_IPV6_LL_ADDR" from NCP: "STATUS_UNIMPLEMENTED" (2)
wpantund[18422]: Network is not joinable
wpantund[18422]: Resetting interface(s). . .
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: Finished initializing NCP
wpantund[18422]: AutoResume is enabled. Trying to resume.
wpantund[18422]: NCP is NOT commissioned. Cannot resume.

**on term 3 Wpanctl**
$ sudo wpanctl 
wpanctl:wpan0> status
wpan0 => [
	"NCP:State" => "offline"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/0.01.00; none; Mar 30 2017 17:33:39"
	"Daemon:Version" => "0.08.00 (/f911961-dirty; Mar 30 2017 21:46:31)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [18B4300000000002]
]
wpanctl:wpan0> scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 1 |       NO | "OpenThread"       | 0x1234 | 11 | DEAD00BEEF00CAFE | B6A5CFFCEB42F575 |  -20
wpanctl:wpan0> join 1 -T 2
Joining "OpenThread" DEAD00BEEF00CAFE as node type "router"
Successfully Joined!

**Logs on terminal 2 observed during scan and joining :**
wpantund[18422]: Scan -> Name:OpenThread       , PanId:0x1234, Ch:11, Joinable:NO , XPanId:0xDEAD00BEEF00CAFE, HwAddr:0xB6A5CFFCEB42F5FE, RSSI:-20 , LQI:0  , ProtoId:3  , Version: 0, ShortAddr:0xFFFF 
wpantund[18422]: State change: "offline" -> "associating"
wpantund[18422]: State change: "associating" -> "associated"
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:302: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/TunnelIPv6Interface.cpp:373: Requirement Failed ((netif_mgmt_add_ipv6_address(mNetifMgmtFD, mInterfaceName.c_str(), addr->__in6_u.__u6_addr8, prefixlen)) == 0)
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:302: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/TunnelIPv6Interface.cpp:373: Requirement Failed ((netif_mgmt_add_ipv6_address(mNetifMgmtFD, mInterfaceName.c_str(), addr->__in6_u.__u6_addr8, prefixlen)) == 0)
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:302: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:355: Requirement Failed (Resource temporarily unavailable)
wpantund[18422]: ../util/netif-mgmt.c:302: Requirement Failed (Resource temporarily unavailable)

**on terminal 3**
wpanctl:wpan0> status
wpan0 => [
	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/0.01.00; none; Mar 30 2017 17:33:39"
	"Daemon:Version" => "0.08.00 (/f911961-dirty; Mar 30 2017 21:46:31)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [18B4300000000002]
	"NCP:Channel" => 11
	"Network:NodeType" => "router"
	"Network:Name" => "OpenThread"
	"Network:XPANID" => 0xDEAD00BEEF00CAFE
	"Network:PANID" => 0x1234
	"IPv6:LinkLocalAddress" => "fe80::10f2:981a:5d2c:8dc9"
	"IPv6:MeshLocalAddress" => "fdde:ad00:beef:0:d5d8:9658:9d63:e1da"
	"IPv6:MeshLocalPrefix" => "fdde:ad00:beef::/64"
	"com.nestlabs.internal:Network:AllowingJoin" => false
]

Opening my Linux Terminal 
by using ifconfig 
**$ ifconfig**
            
wpan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

 **$ ip addr**
 3: wpan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1280 qdisc pfifo_fast state UNKNOWN group default qlen 500
    link/none 

**$ route -6**
Destination                    Next Hop                   Flag Met Ref Use If
2001::/32                               ::                         UAe  256 0     0 eth0
fe80::/64                                ::                         U    256 0     0 eth0
::/0                                         fe80::2e41:38ff:feb2:7ab0  UGDAe 1024 4   347 eth0
::/0                                          ::                         !n   -1  1   602 lo
::1/128                                    ::                         Un   0   2     1 lo
fe80::/128                               ::                         Un   0   1     0 lo
fe80::6600:6aff:fe0b:da31/128  ::                         Un   0   2    10 lo
ff00::/8                                      ::                         U    256 4 23960 eth0
::/0                                            ::                         !n   -1  1   602 lo

from the above info, i see no where NCP mesh local prefix is added.
Now i tried pinging from Linux terminal to Both NCP and CLI Mesh Local addresses ,But it is not working
**from Linux terminal pinging to NCP address**
$ ping6  fdde:ad00:beef:0:d5d8:9658:9d63:e1da

PING fdde:ad00:beef:0:d5d8:9658:9d63:e1da(fdde:ad00:beef:0:d5d8:9658:9d63:e1da) 56 data bytes
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=1 Destination unreachable: Beyond scope of source address
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=2 Destination unreachable: Beyond scope of source address
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=3 Destination unreachable: Beyond scope of source address
^C
--- fdde:ad00:beef:0:d5d8:9658:9d63:e1da ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2016ms

**from Linux terminal pinging to NCP $ ping6 fdde:ad00:beef:0:a7ac:d8b:2b3:456a
PING fdde:ad00:beef:0:a7ac:d8b:2b3:456a(fdde:ad00:beef:0:a7ac:d8b:2b3:456a) 56 data bytes
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=1 Destination unreachable: Beyond scope of source address
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=2 Destination unreachable: Beyond scope of source address
From fe80::2e41:38ff:feb2:7ab0 icmp_seq=3 Destination unreachable: Beyond scope of source address
^C
--- fdde:ad00:beef:0:a7ac:d8b:2b3:456a ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2016ms
address**

So in both cases it is failed. I didn't this kind of Behavior in the previous version of wpantund 0.07.0.
I stuck with this from last two days.
Please suggest me how to get resolved ?

Thanks and Regards
R Madhu.

abtink, 2017-03-30:
Hi Madhu...Thanks for reporting this. I could also reproduce this locally. Will be working to address this...

rmadhuraj, 2017-03-30:
Thank you @abtink , 
I'm pleased to see your updates.
Regards 
R Madhu

darconeous, 2017-03-30:
In the mean time, you may want to try using the `latest-unstable` tag, `full/latest-unstable` tag, or the `release/0.07/master` branch.

abtink, 2017-03-30:
https://github.com/openthread/wpantund/pull/149 should address this. Worked ok on my local testing.

darconeous, 2017-03-30:
The fix has been merged. Reopen if that didn't fix it.

rmadhuraj, 2017-03-31:
HI @darconeous @abtink 

With the Latest Commit , all going well on Posix simulation.
But,
When i do the same things on the Nordic Boards , i'm NOT able to Ping between the NCP and CLI 
and Pinging from Linux terminal to CLI node.
I followed the same procedures mentioned above for compilation, and flashed onto Nordic boards and connected to my Linux PC though USB ports.

**In term 1**
**Through serial console application i configured cli Node,**
panid 0x1234
 Done
ifconfig up
Done
thread start
Done
 state
    leader.
    Done
ipaddr
  fdde:ad00:beef:0:0:ff:fe00:fc00
  fdde:ad00:beef:0:0:ff:fe00:7800 
  fdde:ad00:beef:0:7dd4:f3b2:2dac:a1a9
  fe80:0:0:0:a89d:afe:c741:a253
  Done

and after i did the same as mentioned in the first comment Like scan, joining . . 

**In term 2 Running NCP instance**
$ sudo /usr/local/sbin/wpantund -o NCPSocketNme /dev/ttyACM0 -o WPANInterfaceName wpan0

wpantund[19429]: Starting wpantund 0.08.00 (Mar 31 2017 11:32:13) . . .
wpantund[19429]: 	BUILD_VERSION = f911961-dirty
wpantund[19429]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[19429]: Configuration file "/etc/wpantund.conf" read.
wpantund[19429]: CONFIGURATION PROPERTY "NCPSocketName" IS DEPRECATED. Please use "Config:NCP:SocketPath" instead.
wpantund[19429]: CONFIGURATION PROPERTY "WPANInterfaceName" IS DEPRECATED. Please use "Config:TUN:InterfaceName" instead.
wpantund[19429]: set-config-param: "Config:NCP:SocketPath" = "/dev/ttyACM0"
wpantund[19429]: set-config-param: "Config:TUN:InterfaceName" = "wpan0"
wpantund[19429]: Ready. Using DBUS bus ":1.0"
wpantund[19429]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[19429]: Stat (autolog): Tx history is empty
wpantund[19429]: Stat (autolog): 
wpantund[19429]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[19429]: Stat (autolog): Rx history is empty
wpantund[19429]: Stat (autolog): 
wpantund[19429]: Stat (autolog): NCP state history is empty.
wpantund[19429]: Stat (autolog): 
wpantund[19429]: Stat (autolog): 
wpantund[19429]: Opening tun interface socket with name "wpan0"
wpantund[19429]: Running as root without dropping privileges!
wpantund[19429]: Initializing NCP
wpantund[19429]: NCP is now BUSY.
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_IPV6_ML_ADDR) tid:0
wpantund[19429]: Adding IPv6 Address...
wpantund[19429]: TunnelIPv6Interface: Adding address "fdde:ad00:beef:0:427d:9ce4:ce6f:9563" to interface "wpan0".
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_KEY_SEQUENCE_COUNTER) tid:0
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_IPV6_ADDRESS_TABLE) tid:0
wpantund[19429]: Adding IPv6 Address...
wpantund[19429]: TunnelIPv6Interface: Adding address "fe80::c48e:594d:c9c3:497a" to interface "wpan0".
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_THREAD_LEADER_NETWORK_DATA) tid:0
wpantund[19429]: [-NCP-] Leader network data: 
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_THREAD_ON_MESH_NETS) tid:0
wpantund[19429]: [->NCP] CMD_PROP_VALUE_GET(PROP_PROTOCOL_VERSION) tid:1
wpantund[19429]: 	↳ 810201
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_PROTOCOL_VERSION) tid:1
wpantund[19429]: [->NCP] CMD_PROP_VALUE_GET(PROP_NET_STACK_UP) tid:2
wpantund[19429]: 	↳ 820242
wpantund[19429]: [NCP->] CMD_PROP_VALUE_IS(PROP_NET_STACK_UP) tid:2
wpantund[19429]: State change: "uninitialized" -> "offline"
[... 144 lines cut ...]
3 packets transmitted, 3 received, 0% packet loss, time 1998ms
rtt min/avg/max/mdev = 0.015/0.038/0.059/0.018 ms

its is pingable

But when i try to ping with any of CLI MeshLink local addresses ,its failed
**$ ping6 fdde:ad00:beef:0:7dd4:f3b2:2dac:a1a9**
PING fdde:ad00:beef:0:7dd4:f3b2:2dac:a1a9(fdde:ad00:beef:0:7dd4:f3b2:2dac:a1a9) 56 data bytes
^C
--- fdde:ad00:beef:0:7dd4:f3b2:2dac:a1a9 ping statistics ---
6 packets transmitted, 0 received, 100% packet loss, time 5040ms

Also i tried pinging to NCP from CLI node. It is also failed.
  ..................................................................................................
While testing on POSIX and on BOARDS I observed the only difference is the one command

$ sudo /usr/local/sbin/wpantund -o NCPSocketNme /dev/ttyACM0 -o WPANInterfaceName wpan0

$ sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp-ftd 2" -o SyslogMask " -info" -o Config:TUN:InterfaceName wpan0 

from this , can you guess any changes required in the configuration of Wpantund.conf .? while running on Boards ?

Any idea.?
Thanks and Regards.
R Madhu.

pszkotak, 2017-03-31:
Hi @rmadhuraj,

I have tested your setup (with two NRF52840 DK boards) with openthread commit 959b648 and wpantund commit c8fa12c.
I was able to ping the cli from the linux machine.
Please find logs below.

```
> panid 0x1234
Done
> ifconfig up
Done
> thread start
Done
> ipaddr
fdde:ad00:beef:0:c8d3:d411:2cfc:2f71
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:8800
fdde:ad00:beef:0:139c:ac4a:7c15:692
fe80:0:0:0:a845:b412:ade6:4980
Done

wpanctl:wpan0> scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 1 |       NO | "OpenThread"       | 0x1234 | 11 | DEAD00BEEF00CAFE | AA45B412ADE64980 |  -16
 2 |       NO | "OpenThread"       | 0xCAFE | 13 | DEAD00BEEF00CAFE | 82493FF383F1634F |  -77
 3 |       NO | "OpenThread"       | 0x1234 | 19 | DEAD00BEEF00CAFE | 86F35A733DF836C9 |  -56
wpanctl:wpan0> join 1 -T 2
Joining "OpenThread" DEAD00BEEF00CAFE as node type "router"
Successfully Joined!
wpanctl:wpan0> status
wpan0 => [
	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/g034a210-dirty; NRF52840; Mar 27 2017 08:51:18"
	"Daemon:Version" => "0.08.00 (0.07.01rc1-28-gc8fa12c; Mar 31 2017 13:38:58)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [0761C19F05219B7B]
	"NCP:Channel" => 11
	"Network:NodeType" => "router"
	"Network:Name" => "OpenThread"
	"Network:XPANID" => 0xDEAD00BEEF00CAFE
	"Network:PANID" => 0x1234
	"IPv6:LinkLocalAddress" => "fe80::6088:202f:e861:8c51"
	"IPv6:MeshLocalAddress" => "fdde:ad00:beef:0:67:62f5:56fa:297d"
	"IPv6:MeshLocalPrefix" => "fdde:ad00:beef::/64"
	"com.nestlabs.internal:Network:AllowingJoin" => false
]
wpanctl:wpan0> 

pisz@debian:~$ ping6 fdde:ad00:beef:0:c8d3:d411:2cfc:2f71
PING fdde:ad00:beef:0:c8d3:d411:2cfc:2f71(fdde:ad00:beef:0:c8d3:d411:2cfc:2f71) 56 data bytes
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=1 ttl=64 time=127 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=2 ttl=64 time=42.6 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=3 ttl=64 time=45.9 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=4 ttl=64 time=44.0 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=5 ttl=64 time=43.6 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=6 ttl=64 time=45.9 ms
64 bytes from fdde:ad00:beef:0:c8d3:d411:2cfc:2f71: icmp_seq=7 ttl=64 time=45.0 ms
^C
--- fdde:ad00:beef:0:c8d3:d411:2cfc:2f71 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6008ms
rtt min/avg/max/mdev = 42.639/56.456/127.911/29.194 ms
```
Could you confirm that you are building the same or not older commits?

@darconeous, We have captured the following. Sending TLV with the fdde:ad00:beef::/64 prefix seems to be source of presence of two EIDs(with Mesh Local Prefix) on the CLI. Is it a correct behavior?
[image]

rmadhuraj, 2017-03-31:
Hi @pszkotak ,
       i'm happy that you did it.

   Now I started Testing with the latest commits. Can you Please tell How you compiled the code for Nordic.

I followed the steps below

$ cd openthred-master
$ ./bootstrap
$ make -f examples/Makefile-nrf52840

Is this fine ? Or any configuration required.

Thanks and Regards
R Madhu.

pszkotak, 2017-03-31:
@rmadhuraj,
```
./bootstrap
make -f examples/Makefile-nrf52840
```
is perfectly fine.

regarding the wpantund, I prefer to go with
`./configure --sysconfdir=/etc`
before make and then adjust /etc/wpantund.conf file.
That way you can simply start it by typing `wpantund`.

rmadhuraj, 2017-03-31:
@pszkotak 
        sorry to say again that still i'm facing the same issue after testing with latest commits.
I just now downloaded zip files of openthread and wpantund  and installed the way as you mentioned above.

For configuring the wpantund is this step fine ?

**$ sudo /usr/local/sbin/wpantund -o NCPSocketNme /dev/ttyACM0 -o WPANInterfaceName wpan0**

or do i need to change any parameters manually in the Wpantund.conf file.?

Thanks and Regards
R Madhu.

pszkotak, 2017-03-31:
Parameters you are using are deprecated, but according to the log still correctly interpreted.
Result should be the same.
I use wpantund.conf with the following lines uncommented:
```
Config:NCP:SocketPath "serial:/dev/ttyACM0,raw,b115200,crtscts=1"
Daemon:SyslogMask "all -info -debug"
```

Could you tell a bit more about operating system you are using?
What distribution is it?
It there a firewall enabled?

When reflashing DKs please make sure you use the chiperase switch, eg:
nrfjprog --chiperase --family NRF52 --program <HEX_FILE_HERE> --reset

rmadhuraj, 2017-04-01:
HI @pszkotak .
       My PC is UBUNTU 14.04.
However , i tested in another PC by following your inputs. Now it is pinging in all cases that we discussed ,
Thanks for all your inputs.

And one last question
what is the **tcpdump -i wpan0 ip6** command ?
It is capturing the data only in the case when ping is failed.(i.e, only packets transmitting but failed to receive).

Thanks and Regards.
R Madhu.

darconeous, 2017-04-02:
Reopening so that @abtink has a look.

abtink, 2017-04-02:
@rmadhuraj  do you see the original issue of no address of wpan0 interface?  Seems like you had a follow-up question/issue with Nordic dev board platform...and from your last comment it looks like it's all good.

darconeous, 2017-04-02:
@abtink: Sorry for the lack of clarification. I saw this from @pszkotak:

> @darconeous, We have captured the following. Sending TLV with the fdde:ad00:beef::/64 prefix seems to be source of presence of two EIDs(with Mesh Local Prefix) on the CLI. Is it a correct behavior?

I figured you were in a better position to answer.

Feel free to re-close.

rmadhuraj, 2017-04-03:
yes @abtink , Now everything is working fine.

I want to thank you all the participants once again.

closed 2017-04-03 by rmadhuraj