openthread/openthread #1540: Raw Link Layer Issues on nrf52840

nibanks, 2017-03-30:
I have spent the last two days getting Windows to work again, and I cannot for the life of me figure out what is going wrong on the Nordic Semiconductor devices. When I use two cc2538 devices and connect one to the other, it works just fine. Logs: [win_cc2538.txt](https://github.com/openthread/openthread/files/883844/win_cc2538.txt)

But when I try with two nrf52840 devices, 100% of the time, the Parent Response message fails to get ACK'ed and the as far as I can tell, the other side never has any indication of seeing it. Logs: [win_nrf52840.txt](https://github.com/openthread/openthread/files/883846/win_nrf52840.txt)

```
 7180 3/30/17 13:15:28.581    0004    043C                      [otCore]    [FFFF818A3E760008]MLE Sent parent request to routers
 7310 3/30/17 13:15:28.611    0004    0408                      [otCore]    [FFFF818A3E667008]MLE Received parent request
 7322 3/30/17 13:15:28.695    0004    0408                      [otCore]    [FFFF818A3E667008]MLE Sent delayed response
 7392 3/30/17 13:15:28.744    0004    0408                      [otCore]    [FFFF818A3E667008]MAC Tx failed - Error: Abort (12) - SeqNum: 53 - Attempt 1/4 - Dst: 3e:f8:c8:5c:78:48:53:f8

 7417 3/30/17 13:15:29.331    0004    043C                      [otCore]    [FFFF818A3E760008]MLE Sent parent request to all devices
 7554 3/30/17 13:15:29.361    0004    0408                      [otCore]    [FFFF818A3E667008]MLE Received parent request
 7570 3/30/17 13:15:30.304    0004    0408                      [otCore]    [FFFF818A3E667008]MLE Sent delayed response
 7645 3/30/17 13:15:30.354    0004    0408                      [otCore]    [FFFF818A3E667008]MAC Tx failed - Error: Abort (12) - SeqNum: 54 - Attempt 1/4 - Dst: 3e:f8:c8:5c:78:48:53:f8
```

The logs don't include the device side logs, but those just indicate it continually retries until it gives up. Also, for the nrf52840, I have tried with `OPENTHREAD_CONFIG_ENABLE_PLATFORM_USEC_BACKOFF_TIMER` equal to both `1` and `0` just in case that was the code that broke it. It didn't seem to make a difference.

LuDuda, 2017-03-31:
Thanks for reporting this issue. Can you help me to reproduce this on my local PC? What should i do? I have either Win 7 or Win 10 to test it.

Do you have any sniffer logs? I am wondering if Parent Response or ACK on Parent Response is not sent. What is also strange is that `kThreadError_Abort` is returned, however our driver can only return `kThreadError_None `or `kThreadError_ChannelAccessFailure`, can you add a log of error code in this line: drivers/windows/otLwf/radio.c#L583

nibanks, 2017-03-31:
When I get a chance, I will add the extra log to otLwf. I do not have sniffer logs. To repro, you will need Windows 10 (feel free to setup a test machine with a Windows 10 preview image for free), and you can follow the instructions here:

https://github.com/openthread/openthread/wiki/OpenThread-on%C2%A0Windows

If you have any problems or questions getting set up, feel free to email them to me.

nibanks, 2017-03-31:
As far as why the error was being reported incorrectly at the otLwf layer, it was because of an issue with encoding the error. I have submitted #1544 for a fix. But that obviously doesn't fix the main problem.

LuDuda, 2017-04-02:
I've just tried to follow the instruction on my private Win 10 machine.

After i run `install_otlwf.cmd` and `install_ottmp.cmd` and reboot (also i did `devcon.exe restart *ottmp*`) - i got following error all the time, i tried to reinstall it multiple of times.

```
C:\otwin>otCli.exe
Assertion failed: mApiInstance, file ..\..\src\cli\cli.cpp, line 206
```

Both installation looks correct - no errors or warnings. Where can i found some logs?

nibanks, 2017-04-02:
Did you install the binaries downloaded from AppVeyor (the zip file) or did you build them yourself? And you made sure to enable test signing?

LuDuda, 2017-04-02:
Yes i used the link from the doc (.zip). I did enable test signing by the command in the doc (i had to disable Secure Boot Policy)

But now i enter to Windows test mode by "`Disabling Driver Signature Enforcement`" in Startup Settings, and I can connect to the kit!

I will do some debugs and back to you soon. Thanks!

Side note: I had a lot of blue screens (reason: evsbc8.sys - Driver IRQ Not Less or Equal) when i executed `devcom restart *ottmp*` but i fixed it by disable virtual COM1 and COM2 on my PC.

nibanks, 2017-04-03:
That's great news. It might be useful for you to update the Wiki with any issues and fixes you encountered in your setup. As for the crashes, I'd be interested in taking a look at the dumps. I believe [the following](https://msdn.microsoft.com/en-us/library/windows/hardware/ff542953(v=vs.85).aspx) has the info for collecting the dump. If you get the dump, I will also need to know what version of the thread binaries (which link) you used.

nibanks, 2017-04-10:
Hi @LuDuda. Any progress here?

LuDuda, 2017-04-10:
Was busy last week i did reproduce the issue today and will try to give you some asnwers in following days. Can you guide me how to enable all those logs that you have attached in the first post? Also did you build examples with --enable-link-raw-api or it does not matter?

nibanks, 2017-04-10:
The default for link raw is actually enabled, so you shouldn't have to do anything to enable it. As for the logs, the scripts you run to install the driver also enable all the logging. You can use [tracelog.exe](https://msdn.microsoft.com/en-us/windows/hardware/drivers/devtest/tracelog) to flush the logs to file:

```
tracelog.exe -flush thread
```

That will dump the file name to the console. Then you can convert that file with netsh:

```
netsh trace convert <input> output=thread.txt overwrite=yes report=no tmfpath=<release.zip extracted folder>\symbols\traceformat
```

LuDuda, 2017-04-11:
So i have sent pull request that fixes the problem. Right now i can join to the Thread network successfully.

Btw. I see quite heavy traffic of non-Thread messages in the network, is it expected?

[image]

Thanks for reporting the issue!

nibanks, 2017-04-11:
Thanks for figuring this out! As for Windows being noisy, that is to be expected. A lot of different stuff tries to use the network to discover neighbors.

PR #1580 (2017-04-11): Remove use of CoAP kAckTimeout in Link Raw API.
This PR fixes #1540 

Link Raw have started ACK timer with 2 miliseconds defined by `CoAP::kAckTimeout` instead of 16 miliseconds defined by `Mac::kAckTimeout`.

On nRF52840 it caused aborting of sending MLE Child ID Request which couldn't be transmitted in that short time.

Additionally the root cause has been removed.
closed 2017-04-11 by jwhui