openthread/openthread #1574: Add Bootloader Backdoor to CC2538 Example

CampbellWray, 2017-04-10:
In commit 26e9f3db8b1fc4f60cac890414637e75c8aada18 the bootloader backdoor was added to the CC2650 Example. 

Could a similar backdoor please be added to the CC2538 example

jwhui, 2017-04-10:
I believe the backdoor is already enabled in the CC2538 example.  See the lock page configuration in [examples/platforms/cc2538/startup-gcc.c](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2538/startup-gcc.c#L158).

CampbellWray, 2017-04-11:
Looks like you might be right, it might be worth updating the readme (similar to the CC2650 one) so that it is more obvious that it is there and how to use it

jwhui, 2017-04-11:
@CampbellWray, thanks for the suggestion.  I just submitted #1582 to update the README.

CampbellWray, 2017-04-12:
Looks good, I'll give it a try when I next have the CC2538s in front of me

PR #1582 (2017-04-12): Update CC2538 flashing instructions.
Resolves #1574.
closed 2017-04-12 by CampbellWray