openthread/openthread #1593: COAP server and client not working with POSIX

CampbellWray, 2017-04-14:
When compiling the POSIX version of OpenThread the `coap server` and `coap client` commands do not work.

The commands do not show up with the `help` command either.

I have the latest version of the source code downloaded, and I am building it using:
```
make -f examples/Makefile-posix
```
Am I missing a necessary flag to get this to compile properly, or are the COAP commands just incompatible with the POSIX version?

jwhui, 2017-04-14:
Try adding `--enable-application-coap` to the `configure_OPTIONS` list in [examples/Makefile-posix](https://github.com/openthread/openthread/blob/master/examples/Makefile-posix#L84).

CampbellWray, 2017-04-14:
I tried that earlier and I got the following error:
```
Undefined symbols for architecture x86_64:
  "Thread::Cli::Coap::Process(otInstance*, int, char**, Thread::Cli::Server&)", referenced from:
      Thread::Cli::Interpreter::ProcessCoap(int, char**) in libopenthread-cli.a(libopenthread_cli_a-cli.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [ot-cli-mtd] Error 1
make[5]: *** Waiting for unfinished jobs....
Undefined symbols for architecture x86_64:
  "Thread::Cli::Coap::Process(otInstance*, int, char**, Thread::Cli::Server&)", referenced from:
      Thread::Cli::Interpreter::ProcessCoap(int, char**) in libopenthread-cli.a(libopenthread_cli_a-cli.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
However using a different machine, where the architecture is `i686` it works without any problems.

Perhaps the issue is specific to some architectures

jwhui, 2017-04-14:
I'll take a look.

jwhui, 2017-04-14:
I'm not able to reproduce your issue.  Can you provide more information about your environment?

```bash
$ uname -a
Linux jonhui-virtual-machine 4.8.0-41-generic #44~16.04.1-Ubuntu SMP Fri Mar 3 17:11:16 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
```

CampbellWray, 2017-04-15:
It is running macOS so I imagine that is the issue, it is odd because I haven't had any previous issues 
```
$ uname -a
Darwin machines-name 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64
$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.41)
Target: x86_64-apple-darwin16.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jwhui, 2017-04-15:
Have you tried a clean rebuild?

```bash
git clean -xfd
./bootstrap
make -f examples/Makefile-posix
```

CampbellWray, 2017-04-15:
That seemed to help, but it is now getting a different error
```
/Users/Username/Documents/Git/openthread/examples/../src/cli/cli_coap.cpp:83:22: error: implicit
      conversion loses integer precision: 'int' to 'char' [-Werror,-Wconversion]
        aString[i] = tolower(aString[i]);
                   ~ ^~~~~~~~~~~~~~~~~~~
1 error generated.
```

jwhui, 2017-04-15:
Yup, I'm waiting to merge #1595 before submitting the fix, which is to cast to chat.

CampbellWray, 2017-04-15:
Great, I have managed to temporarily fix it on my end by adding a char cast, would you like me to keep this issue open, or is this already covered elsewhere?

jwhui, 2017-04-15:
Keep this open.  I'll tag this issue in the PR so that it automatically closes after merge.  Thanks for surfacing these issues!

CampbellWray, 2017-04-15:
I thought I had better ask here before opening another issue as it is somewhat related.

It could simply be that I am doing something wrong but the coap client command seems to be causing segmentation faults.

For example:
```
coap client put fdde:ad00:beef:0:9687:6ac:fb08:7340 test con 123Segmentation fault: 11
```
I could just be using it wrong but either way, it might be worth putting something in place to stop this from happening.

jwhui, 2017-04-17:
Just submitted #1605.  Please review.

PR #1605 (2017-04-18): Fix compiler warning and code style.
Resolves #1593.
closed 2017-04-18 by jwhui