openthread/openthread #1601: coap client command causing segmentaion fault

CampbellWray, 2017-04-15:
I have had a look around and I believe that this could be due to the recent changed to the COAP API in commit b6e72560a9cf73dd47a7d6a36fc7c827bfd7d91e.

When running the `coap client ...` command in the POSIX application it will crash.

I have run it through gdb and this was the output:
```
(gdb) run
Starting program: /home/user/Documents/openthread/output/i686-pc-linux-gnu/bin/ot-cli-ftd 1
> ifconfig up
Done
> ipaddr
fdde:ad00:beef:0:9832:4707:d09c:9a7d
fe80:0:0:0:48ed:5826:6782:8615
Done
> coap server start
Server started with resource '': Done
> coap client put fdde:ad00:beef:0:9832:4707:d09c:9a7d test con Message
                                                                       Program received signal SIGSEGV, Segmentation fault.
                                                                                                                           Thread::Coap::CoapBase::NewMessage (this=0x80cfffc <sInstanceRaw+13948>, aHeader=..., aPriority=2 '\002')
    at /home/user/Documents/openthread/examples/../src/core/coap/coap_base.cpp:45
45	    VerifyOrExit(aHeader.GetLength() >= Header::kMinHeaderLength);
(gdb) where
#0  Thread::Coap::CoapBase::NewMessage (this=0x80cfffc <sInstanceRaw+13948>, aHeader=..., aPriority=2 '\002')
    at /home/user/Documents/openthread/examples/../src/core/coap/coap_base.cpp:45
#1  0x0806af57 in otCoapNewMessage (aInstance=0x80cc980 <sInstanceRaw>, aHeader=0x0) at /home/user/Documents/openthread/examples/../src/core/api/coap_api.cpp:131
#2  0x080503d9 in Thread::Cli::Coap::ProcessClient (argc=5, argv=0xbfffecc0) at /home/user/Documents/openthread/examples/../src/cli/cli_coap.cpp:259
...
```

jwhui, 2017-04-17:
Thanks for posting this issue.  Couple issues while looking into this:

1. [otCoapNewMessage()](https://github.com/openthread/openthread/blob/master/src/core/api/coap_api.cpp#L129) requires `aHeader` to be non-NULL but does not actually validate the argument.
2. The first call to `otCoapNewMessage()` in [Coap::ProcessClient](https://github.com/openthread/openthread/blob/master/src/cli/cli_coap.cpp#L259) should be removed and replaced with `NULL`.  `otCoapNewMessage()` should be called after the header is created, which is done correctly later in the same method.

Would you be willing to submit a PR that fixes these two issues?

CampbellWray, 2017-04-17:
Okay so changing `line 259` to read `otMessage *message = NULL;` will stop the message from being declared with a NULL header (therefore fixing the segmentation fault).

Other than that I can't see any changes that need to be made.

On `line 321` the header is initialised by passing the `coapType` and `coapCode` to `otCoapHeaderInit`, both of which will both be non-NULL, so we can be certain that the header would be non-NULL and the message can be declared safely on `line 326`.

From that, I can only see one change, but if that was all that needs doing I will happily submit a pull request with the change

CampbellWray, 2017-04-17:
Unless you are talking about adding a validation check into the API code as well to make sure that it can't be initialised with NULL in the future

jwhui, 2017-04-17:
Yes, I am proposing that both changes should be made :)

CampbellWray, 2017-04-17:
Right, I re-read it afterward and realised what you were saying, shouldn't be a problem, I'll get onto that now

CampbellWray, 2017-04-17:
I've had a look into this, I should probably start by saying that I have never worked with c++ before, but I am still more than willing to fix this problem. 

Starting with `otCoapNewMessage` It calls `mThreadNetif.GetCoapClient().NewMessage(*(static_cast<const Coap::Header *>(aHeader)));` so I followed the files that goes through:

1. `thread/thread_netif.cpp` uses the `GetCoapClient` method defined in `thread_netif.hpp` this leads to...
2. `coap/coap_client.cpp` but there is no reference to a `NewMessage` method.
3. I assume from here it must use `NewMessage` from `coap_base.cpp` (not sure how it does this but its the only place NewMessage is defined, perhaps through the `#include` but not sure).

This matches the gdb snippet I posted earlier where the crash appears to occur when `aHeader.GetLength()` is called, I assume because `aHeader` is `NULL` so it is calling `NULL.GetLength()`.

Here is would make sense to verify that `aHeader` is not `NULL` before `.GetLength()` is called, currently I am using `VerifyOrExit(aHeader != NULL);` .

I am now in a situation where the code won't compile, I am wondering if I should be checking `aHeader` itself, pointer to (`*aHeader`) it or the memory address of it (`&aHeader`).

Finally I am pretty sure that I am in the right place, however it would be good to get clarification on that.

Otherwise if anything I have said doesn't make sense of if you have any other suggestions would be great.

jwhui, 2017-04-17:
The NULL-pointer check should go in C-API `otCoapNewMessage()` wrapper in [src/core/api/coap_api.cpp](https://github.com/openthread/openthread/blob/master/src/core/api/coap_api.cpp#L131).  The internal call to `NewMessage()` takes a reference, which implies that NULL is not a valid value.

CampbellWray, 2017-04-18:
Okay so looking at how`message_api.cpp` handles errors, I have gone ahead and done this:
```
otMessage *otCoapNewMessage(otInstance *aInstance, const otCoapHeader *aHeader)
{
    Message *message;
    
    VerifyOrExit(aHeader != NULL, message = NULL);
    message = aInstance->mThreadNetif.GetCoapClient().NewMessage(*(static_cast<const Coap::Header *>(aHeader)));
    
exit:
    return message;
}
```
Is this acceptable, by returning a NULL message it's going to likely cause errors elsewhere but maybe that should be left for the code using the API to check whether the message was created successfully.

jwhui, 2017-04-18:
Yes, that is perfectly fine.  NULL may also be returned if there are no message buffers available.  As a result, the caller needs to check the return value anyway.

CampbellWray, 2017-04-18:
#1606 Submitted

PR #1606 (2017-04-18): Coap fixes
Resolves #1601 and creates adds check for NULL header in the API to avoid a similar issue in the future
closed 2017-04-18 by jwhui