openthread/openthread #1607: API Reference page 404

CampbellWray, 2017-04-18:
Link to the API Reference page (https://openthread.github.io/openthread/) is giving a 404 error, and has been for the last day or so, is this just me or is this affecting others too?

If it is just me, is there somewhere else that these documents are hosted?

jwhui, 2017-04-18:
The existing API reference has been removed for the following reasons:

1. The API reference contained many images that were bloating this repo.
2. In part due to the prior reason, the API reference was horribly out-of-date.

We are looking for a new site to host the API reference.  In the mean time, you can install Doxygen and Graphviz.  Once installed, the docs will generate automatically as part of your build.

closed 2017-04-19 by jwhui

jwhui, 2017-05-10:
@CampbellWray, the API reference is now hosted at: https://openthread.github.io/openthread-docs/