openthread/openthread #1611: Provisining support

GiedriusM, 2017-04-18:
To my understanding provisioning currently is not implemented in the OpenThread stack. I wanted to know if this feature is planned for the near future and create an active issue as a feature request if it isn't.

I do realize, that provisioning content is application specific, however at least some API and/or hooks should be available for the application as JOIN_APP commands are a part of Thread spec.

[OFFTOPIC]
Also as a side comment, Thread spec states that provisioning is done when device is attached to the network (8.9.3#33), that is, after finalize and entrust. The joiner and the commissioner (provisioner?) app "synchronizes" using provisioning URL TLV in JOIN_APP command, the commissioning itself has no (does not require any) vendor specific parts and any joiner device could be commissioned to the network using any generic app. Then what is the point of having provisioning information (URL) in JOIN_FIN commands? Is it only to keep the DTLS session alive?
I would say that what many applications/developers could do is to not include the provisioning URL when commissioning, so that any app could be used to join the network. Once the device is on the network, new DTLS session would be made and the vendor provisioning URL used in JOIN_APP.req. Or even a custom service discovery could be used (although I do believe that provisioning is a better choice).

jwhui, 2017-04-18:
Application-layer provisioning through the MeshCoP protocol defined in Section 8.9.3 of the Thread Specification is not implemented at this time.  This is an enhancement we can look into if requested.  As always, we welcome contributions as well.

The provisioning URL information provides a means to inform the Commissioner of an application-specific Commissioner application desired by the Joiner.  The JOIN_FIN.rsp message allows the Commissioner to inform the Joiner whether it has the correct Commissioner application.

As you noted, using the same DTLS session to provision both the Thread and application layers is optional.  Reusing the DTLS session provides the following abilities:

1. Reduce overhead associated with establishing a separate DTLS session for the application layer.
2. Allow the joining device to choose not to participate in the Thread network if the application layer cannot be provisioned.

That said, there are many good reasons to have the device join the network then attempt to provision the application layer separately.  Ultimately, the choice is application/solution specific.

jwhui, 2021-04-07:
Closing stale issue. If this is still of interest, feel free to reopen.

closed 2021-04-07 by jwhui